stepjam/PyRep #121: Error with `/examples/example_reinforcement_learning_env`

loicsacre, 2020-02-10:
I am getting this error when launching `/examples/example_reinforcement_learning_env.py` (no modifications were applied):

```
Initialization successful.
Starting episode 0
Traceback (most recent call last):
  File "example_reinforcement_learning_env.py", line 81, in <module>
    state = env.reset()
  File "example_reinforcement_learning_env.py", line 47, in reset
    return self._get_state()
  File "example_reinforcement_learning_env.py", line 40, in _get_state
    self.target.get_position())
ValueError: operands could not be broadcast together with shapes (14,) (3,) 
```

Info:
- Ubuntu 16.04
- Python3.7
- CoppeliaSim_Edu_V4_0_0_Ubuntu16_04

loicsacre, 2020-02-10:
Maybe, the return type of `get_position(self, relative_to=None)` in `objects/object.py` should be a List of float ?

stepjam, 2020-02-10:
Thanks for pointing out :)

commit e1ca4af6 (2020-02-10): Modify example due to changes to api. Resolves #121.
closed 2020-02-10 by stepjam