stepjam/PyRep #129: Access to camera object in the scene.

mch5048, 2020-02-14:
Hi, thanks for this amazing research kit! 

Currently, I'm developing a VR interface written in Python that renders the camera view (not Visionsensor) to the Hmd of VR device. In order to do so, I'm thinking of acquiring the high-resolution video from the camera object in the scene as shown in the figure below.

I guess I can access the camera object in the scene using the Object class you implemented, but can't sure if I can get the high-resolution video.

Or, If I can mimic the camera's FOV and resolution using the vision sensor class, the problem would become easier. (Is there any reference to those parameters of the default camera in CoppeliaSim?)

I want to get some advice from you about the problem I'm facing!

 Thanks in advance!

Best regards,

Cheolhui.
[image]

Coppelia, 2020-02-26:
Hello Cheolhui,
sorry about the late reply. You can't really access the image content of a camera object in CoppeliaSim. Only vision sensor's image content can be accessed. Additionally, not the camera is associated with specific parameters such as FOV, resolution, far/near clipping planes, prespective/orthogonal projection, but the view the camera is associated with (e.g. a camera can have two different views associated, one showing the view in perspective projection mode, the other one in orthogonal projection mode).
Best is really to use a vision sensor. Vision sensor's parameters can mainly be accessed via [object parameter IDs](https://www.coppeliarobotics.com/helpFiles/en/objectParameterIDs.htm)

mch5048, 2020-02-28:
@Coppelia Thank you for your kind reply! Much appreciate your help.

closed 2020-02-28 by mch5048