stepjam/PyRep #131: Syntax error with PyRep package

helenrensiyu, 2020-02-15:
Hi, I just downloaded CoppeliaSim and installed PyRep today. However, when I was trying to run examples, I have gotten the following error:

> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/__init__.py", line 3, in <module>
>     from .pyrep import PyRep
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/pyrep.py", line 1, in <module>
>     from pyrep.backend import sim, utils
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/backend/utils.py", line 8, in <module>
>     from pyrep.objects.object import Object
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/objects/__init__.py", line 1, in <module>
>     from pyrep.objects.object import Object
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/objects/object.py", line 9
>     object_type_to_class: Dict[ObjectType, Any] = {}
>                         ^
> SyntaxError: invalid syntax

In addition to this, while running the installation, other syntax errors were detected too:

> [byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/gripper.py to gripper.cpython-35.pyc
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/gripper.py", line 22
>     self._old_parents: List[Object] = []
>                      ^
> SyntaxError: invalid syntax
> 
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/locobot_gripper.py to locobot_gripper.cpython-35.pyc
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/baxter_gripper.py to baxter_gripper.cpython-35.pyc
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/panda_gripper.py to panda_gripper.cpython-35.pyc
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/jaco_gripper.py to jaco_gripper.cpython-35.pyc
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/baxter_suction_cup.py to baxter_suction_cup.cpython-35.pyc
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/mico_gripper.py to mico_gripper.cpython-35.pyc
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/dobot_suction_cup.py to dobot_suction_cup.cpython-35.pyc
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/suction_cup.py to suction_cup.cpython-35.pyc
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/robots/end_effectors/suction_cup.py", line 17
>     self._old_parents: List[Object] = []
>                      ^
> SyntaxError: invalid syntax
> 
> byte-compiling /home/helen/.local/lib/python3.5/site-packages/pyrep/robots/configuration_paths/arm_configuration_path.py to arm_configuration_path.cpython-35.pyc
>   File "/home/helen/.local/lib/python3.5/site-packages/pyrep/robots/configuration_paths/arm_configuration_path.py", line 23
>     self._rml_handle: Optional[int] = None
>                     ^
> SyntaxError: invalid syntax

I am using Python 3.5.2. Could this be due to different versions of Python? It would be great if someone who tried to install recently or just anyone could shed some light on this. Thank you!!

gesheng199832, 2020-03-14:
https://docs.python.org/3.6/whatsnew/3.6.html#pep-526-syntax-for-variable-annotations

themathgeek13, 2020-03-20:
It's pretty easy to fix these errors for Python3.5, as mentioned by @gesheng199832 above, you can just replace all of the lines which have issues like this:

> self._rml_handle: Optional[int] = None

And replace it with standard Python syntax as shown below:

> self._rml_handle = None  # Optional[int] 

(comment is just to specify the type that was made more explicit in Python3.6)

closed 2020-07-04 by stale[bot]