stepjam/PyRep #132: [ERROR] Adding new robot error

kuonangzhe, 2020-02-16:
Thanks for the great work of PyRep!

I recently wanted to add a new 7DOF robot into PyRep robot pool, by following the `adding new robot` tutorial. I loaded robot urdf model via Urdf plugin and changed to MyRobot_xxx naming, added two dummies tip and target on tip and root, and moved two dummies to the same pose, with z axis pointing out. Also, I added collection and IK, as well as python arm class. The robot also passed the unit test (modified and added the robot into the test file) mentioned in README.

However, when testing by using a simple movement,  `examples/example_panda_end_effector_control.py`, where I changed panda class into MyRobot, in Vrep, I just saw only one of tip/target dummy moves and robot itself doesn't move.  Comparing to the panda example, panda robot moves and dummies seem to keep their pose.

Then it alarms an error at first robot run line, saying `IK failed. Perhaps the distance was between the tip and target was too large' from file `PyRep/pyrep/robots/arms/arm.py`.
```
        ik_result, joint_values = sim.simCheckIkGroup(
            self._ik_group, [j.get_handle() for j in self.joints])
        if ik_result == sim.sim_ikresult_fail:
            raise IKError('IK failed. Perhaps the distance was between the tip '
                          ' and target was too large.')
        elif ik_result == sim.sim_ikresult_not_performed:
            raise IKError('IK not performed.')
```
I think there must be somewhere I configured wrong but I have no idea whats the exact reason. 

Are there any suggestions? Thanks a lot!

kuonangzhe, 2020-02-16:
@stepjam

stepjam, 2020-03-25:
Hi, sorry for dealy.

Rather than using `agent.solve_ik(pos, quaternion=quat)`, try using `agent.get_path(position=pos, quaternion=quat)`.
As the error is saying, the distance may be too large and you instead need to use configurations instead.

stepjam, 2020-03-25:
Or just having the target much closer.
Maybe look at using the `examples/example_panda_reach_target.py` example.

closed 2020-07-04 by stale[bot]