stepjam/PyRep #137: Headless mode, simulation doesn't run in expected behaviour.

Fzaero, 2020-02-25:
Hello, I have been using Pyrep some time, I'm having problem with headless mode. My main problem is that simulation behaves differently when I use it in headless mode. I tried two different computers with ubuntu 18.04, one of them were newly installed ubuntu.

I'm trying to run a water pouring scenario. For this, I first fill a cup with marbles, then pour these marbles to another cup. I added two force sensors before cups which shows before and after force values of cups. I take mean of last 30 values of force values(directly defined in model side). 

I added scene, model files and script files. Scenario is like this, I first run pour_water_demonstration.py to record joint angles of robot. then I run pour_water_run_traj.py or pour_water_run_traj_headless.py to rerun this trajectory. This is force value I read when I run in normal mode.

**pour_water_run_traj.py**
Starting Force Table:  0.49050018191337585  Gripper  26.034360885620117
After Action Force, Table:  5.575415134429932  Gripper  20.89177894592285

**pour_water_run_traj_headless.py**
Starting Force Table:  0.49050018191337585  Gripper  25.433317184448242
After Action Force, Table:  0.4299725890159607  Gripper  23.99286651611328

In second scenario we can see that only few marbles are poured from cup and none of them dropped are actually poured to other cup.

I actually saw a better behaviour on system when I added time.sleep(0.004) after each pr.step() but that makes the system same speed as normal version and probably something I shouldn't do.

**pour_water_run_traj_headless.py with time.sleep(0.004) after pr.step()** 
Starting Force Table:  0.49050018191337585  Gripper  25.3530330657959
After Action Force, Table:  5.974010944366455  Gripper  20.50100326538086

I wanted to ask what should I do in this case. I used
 xvfb-run when I run system headless.

[water_pouring_experiment.zip](https://github.com/stepjam/PyRep/files/4250311/water_pouring_experiment.zip)

- UR10_pouring.ttt : main scene file
- UR10.ttm : I'm adding model file later since I've found that there is an accumulating error after resetting the scene.
- table_with_glass.ttm: Table with glass on it.
- pour_water_demonstration.py: Code that collects an expert demonstration.
- pour_water_run_traj.py: Rerunning expert trajectory on robot.
- pour_water_run_traj_headless.py: Rerunning expert trajectory on robot but headless.

stepjam, 2020-03-26:
Hi,
There shouldn't be any difference between running windowed or headless. I would imagine the differences you are seeing are to with the noise in the physics engine calculations. 
Rather than comparing headless vs windowed, try comparing a few runs runs of a windowed run, and I would imagine you get somewhat similar (non-deterministic) behaviour.
Best,
Stephen

Fzaero, 2020-03-26:
I have tried many times, but this somehow happens all the time.

For example when I created a simple push scene, where robot pushes an object on table, in windowed mode, it pushed the object as expected, but in headless, it bumps the object very fast and throw it of the table. 

I used position control in this scenes, maybe that is causing issue? In the example scripts provided, I see that mostly velocity control is used.

Best,
Ahmet

stepjam, 2020-03-26:
Hi,

I just rang your code and it seems to run as expected... Here is my output:

```
# Recorded
Starting Force Table:  0.49050018191337585  Gripper  26.028215408325195
After Action Force, Table:  6.3793463706970215  Gripper  20.110450744628906

# Playback Windowed
Starting Force Table:  0.49050018191337585  Gripper  25.996183395385742
After Action Force, Table:  6.372431755065918  Gripper  20.110523223876953

# Playback Headless
Starting Force Table:  0.49050018191337585  Gripper  25.996183395385742
After Action Force, Table:  6.372431755065918  Gripper  20.110523223876953
```
One thing I did notice was that your scene file (.ttt) was from an old version of CoppeliaSim. This may be your problem. Ensure that you have the most up-to-date version of CoppeliaSim and then make sure you create the scene in this new version; which in your case is an empty scene anyway. 

Please reopen if there is anything else :) 

-Stephen

closed 2020-03-26 by stepjam

Fzaero, 2020-03-27:
Yes, actually my version was up to date, but I was using a scene file I opened at vrep 3.5(I tried many things to solve problem, but changing the scene file never occurred to me ). Just changing it fixed the problem.

Thank you