stepjam/PyRep #142: missing video compression library

MMSIbrahim, 2020-03-03:
whenever i use coppeliasim through pyrep it gives me the missing video compression library bug. even though coppeliasim works normally on its own and the libraries are all installed.
[image]

Ramomrito, 2020-03-21:
Hi @MMSIbrahim,

I get the same issue when I am using a conda environment. However, when I switched to using a regular virtualenv and installed pyrep there it works fine. I'm guessing it has to do with the anaconda environment not being able to access the installed video compression library.

MMSIbrahim, 2020-03-27:
Hello, @Ramomrito ,

Is there anyway to make it work with anaconda? Let me know if you figure something out.
And thanks for the tip on venv instead of anaconda :)

adamconkey, 2020-04-01:
I have a fix for this, I had the same issue in Ubuntu 18.04 using conda. For me the root cause was `libvvcl.so` (responsible for video compression) in the CoppeliaSim folder was messing everything up. My solution was to build it from source as follows:

    git clone https://github.com/CoppeliaRobotics/videoRecorder.git
    cd videoRecorder/vvcl
    mkdir build
    cd build
    cmake ..
    make

This will create a file `libvvcl.so` in the build directory you just made. Copy that file to the folder where your CoppeliaSim code is (there will already be a version of that file there, just replace it). That's all I had to do.

I came to this solution based on [this thread](https://forum.coppeliarobotics.com/viewtopic.php?f=9&t=7833), if this doesn't work you can try some other things mentioned there. For example, when I did `ldd libvvcl.so` I noticed that it wasn't resolving my system dependencies for `libavcodec*`, `libavformat*`, `libavutil*` and `libswscale*`. I tried as the thread suggests to just remove all the versions of those files in the CoppeliaSim folder, which did allow it to resolve to my system dependencies, but the issue still remained. Building the video recorder and copying the shared object file over was the only thing that worked for me.

Ramomrito, 2020-04-01:
Hi @adamconkey,

Thanks for the help! I tried building the video recorder package and replacing the shared library in the CoppeliaSim folder and it works!

closed 2020-07-04 by stale[bot]

FeiZhou02, 2025-04-24:
> 我有一个解决方案，我在 Ubuntu 18.04 上使用 conda 时也遇到了同样的问题。对我来说，根本原因是`libvvcl.so`CoppeliaSim 文件夹中（负责视频压缩的文件夹）的内容搞乱了一切。我的解决方案是从源代码构建它，如下所示：
> 
> ```
> git clone https://github.com/CoppeliaRobotics/videoRecorder.git
> cd videoRecorder/vvcl
> mkdir build
> cd build
> cmake ..
> make
> ```
> 
> 这将在您刚刚创建的构建目录中创建一个文件`libvvcl.so`。将该文件复制到您的 CoppeliaSim 代码所在的文件夹（那里已经有该文件的版本，只需替换即可）。这就是我要做的。
> 
> 我根据[这个帖子](https://forum.coppeliarobotics.com/viewtopic.php?f=9&t=7833)找到了这个解决方案，如果这个方法不管用，你可以尝试那里提到的其他方法。例如，当我尝试这样做时，我`ldd libvvcl.so`注意到它没有解析我的系统对、和的依赖关系`libavcodec*`。我按照帖子的建议尝试删除 CoppeliaSim 文件夹中这些文件的所有版本，这确实让它解析了我的系统依赖关系，但问题仍然存在。构建视频录像机并复制共享对象文件是唯一对我有用的方法。`libavformat*``libavutil*``libswscale*`

Thank you for providing the methods, but I encountered a failure while compiling VVCL on Ubuntu 24.04.
error: ‘AVPicture’ does not name a type; did you mean ‘AVPictureType’?
  AVPicture PictureIn;

  AVPictureType
error: ‘avcodec_register_all’ was not declared in this scope
   avcodec_register_all();

FeiZhou02, 2025-04-24:
> 每当我通过 pyrep 使用 coppeliasim 时，它都会给我缺少的视频压缩库错误。即使 coppeliasim 本身可以正常工作并且所有库都已安装。 更多图片：[链接](https://imgur.com/a/qpZ0IZB) 我从 2 周前开始使用 Ubuntu 18.04 coppeliasim 4.0 和 pyrep。还使用 anaconda 来编译 python。 [image][](https://imgur.com/a/qpZ0IZB)

I'm also encountering the same issue on Ubuntu 24. How did you resolve it?

whiteHuskie, 2025-04-24:
The method works for Ubuntu 20.04.

FeiZhou02, 2025-04-24:
> 该方法适用于 Ubuntu 20.04。

Thank you for your reply. please，how can i solve this issue to use CoppeliaSim 4.1 on Ubuntu 24?
If I ignore this issue, I won't be able to successfully run RLbench. I tried both CoppeliaSim 4.7 and CoppeliaSim 4.9, but neither worked.

FeiZhou02, 2025-08-14:
> [@FeiZhou02](https://github.com/FeiZhou02) any updates? having same issue on Ubuntu 24

When I changed to a higher version of python, it worked fine.  I refer to this 
https://blog.csdn.net/m0_55930133/article/details/147615429?spm=1001.2014.3001.5501