stepjam/PyRep #145: Solve_IK problem in end effector pose control example

HeegerGao, 2020-03-13:
Hello, I am using the example code 'example_panda_end_effector_control.py' and I replace the Panda arm with Jaco arm. When I try to run the script, I got error like this:

```
Traceback (most recent call last):
  File "pr_test.py", line 30, in <module>
    [move(2, -DELTA) for _ in range(20)]
  File "pr_test.py", line 30, in <listcomp>
    [move(2, -DELTA) for _ in range(20)]
  File "pr_test.py", line 25, in move
    new_joint_angles = agent.solve_ik(pos, quaternion=quat)
  File "/home/gck/.local/lib/python3.6/site-packages/pyrep/robots/arms/arm.py", line 130, in solve_ik
    raise IKError('IK failed. Perhaps the distance was between the tip '
pyrep.errors.IKError: IK failed. Perhaps the distance was between the tip  and target was too large.
QObject::~QObject: Timers cannot be stopped from another thread
QMutex: destroying locked mutex
```
 
pr_test.py is the python file.

I am using vrep4.0. Can you tell me how to fix it? I don't know if the error is caused by my model file. [Here is my .ttt file](https://cloud.tsinghua.edu.cn/f/6df2ccc097ea4326a702/?dl=1), which contains a jaco arm copied from [RLbench repo](https://github.com/stepjam/RLBench/blob/master/rlbench/robot_ttms/jaco.ttm). Thanks!

yzlc080733, 2020-03-13:
I encountered the same issue. My solution is to lower the end effector beforehand through path planning:
`pos[2] -= 0.2`
`agent.get_path(pos, quat)`
Maybe there is difficulty in calculating IK, as the initial position of the end effector is too high?

HeegerGao, 2020-03-13:
> I encountered the same issue. My solution is to lower the end effector beforehand through path planning:
> `pos[2] -= 0.2`
> `agent.get_path(pos, quat)`
> Maybe there is difficulty in calculating IK, as the initial position of the end effector is too high?

Wow! It works! You are so amazing...

closed 2020-03-13 by HeegerGao