stepjam/PyRep #148: Segmentation Fault when running pr.launch

yf291115925, 2020-03-19:
Environment: Ubuntu 16.04 x64 + Anaconda (Python 3.6.5)
Everything works well unitil running an example.
```
yue@yue:~/Env/PyRep/examples$ python example_baxter_pick_and_pass.py 
lib: 1
lic: 1

CoppeliaSim Edu V4.0.0. (rev. 4)
Using the default Lua library.
Segmentation Fault (Core Dump)
```
After some debugging, I found that it crashed when running the pr.launch(...) code.
But when I set the headless mode to True, it works.
So I wonder maybe it's the GPU driver's problem.
I'm using Nvidia GPU GTX1060 with driver 384.
I've also tried Nouveau display driver, but it didn't work.

Could anyone help me? Thanks!

stepjam, 2020-03-26:
Does everything runs fine when you launch CoppeliaSim from command line? i.e. by running `coppeliaSim.sh`. If so, then it isn't a problem with you GPU drivers.
It could be a problem with anaconda. Have you seen:
https://github.com/stepjam/PyRep/pull/93/files

pandyarohan, 2020-03-31:
I am seeing the exact same error, I tried running the shell file recommended above and CoppeliaSim starts up without any issues. I checked all the pointers in the link above, but that doesn't seem to be issue as well. Any other pointers?

yf291115925, 2020-04-02:
Thank you all!
1. Running `coppeliaSim.sh` is ok, so it could be a problem with anaconda. @stepjam 
2. I tried using anaconda3-5.3.1 which includes Python3.7, and followed the steps in #93 . But the same segmentation fault occurred.
3. I uninstallled anaconda, then install python3.7. Everything is good just following README.md. @pandyarohan

closed 2020-04-02 by yf291115925