stepjam/PyRep #149: How to set control frequency in PyRep

HeegerGao, 2020-03-21:
Hello, I am using PyRep for jaco arm control in vrep4.0. I want to know how to set the frequency of the control loop. For example, the get_path function for the arm path planner. If I want to set it to 20hz, how can I do it? Thank you!

HeegerGao, 2020-03-21:
I find a function: `set_simulation_timestep()`.
I think this is what I want, and when I use `get_simulation_timestep()` I get 0.05, which coresponds to 20Hz. But when I reckon the time length between two `pr.step()` in the following code: 
```
path = agent.get_path(position=pos, euler=[0, math.radians(180), 0])
start_time = time.time()
done = False
while not done:
    done = path.step()
    pr.step()
    print('time length:', time.time() - start_time)
    start_time = time.time()
```
I get time length around 0.006, which is much quicker than 0.05. I am very confused.

stepjam, 2020-03-26:
Hi,

The `simulation_timestep` is the timestep used in the **physics calculations**. When you time the `step` function, what you are actually doing is timing how long it takes to done one physics step (where `dt=0.05`).  So if you want to control at 20Hz, then you are already doing the right thing :+1:

closed 2020-03-26 by stepjam