stepjam/PyRep #150: EGLFS cannot open shared object file: No such file or directory

bycn, 2020-03-24:
Trying to use PyRep with EGLFS on docker. I get an error that it can't find a certain file, the full traceback with debug on is below:
```

Got keys from plugin meta data ("offscreen")
QFactoryLoader::QFactoryLoader() looking at "/app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/platforms/libqvnc.so"
Found metadata in lib /app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/platforms/libqvnc.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "vnc"
        ]
    },
    "archreq": 0,
    "className": "QVncIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("vnc")
QFactoryLoader::QFactoryLoader() looking at "/app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/platforms/libqxcb.so"
Found metadata in lib /app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "archreq": 0,
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/root/miniconda3/envs/rlkit/bin/platforms" ...
Cannot load library /app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/platforms/libqeglfs.so: (libQt5EglFSDeviceIntegration.so.5: cannot open shared object file: No such file or directory)
QLibraryPrivate::loadPlugin failed on "/app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/platforms/libqeglfs.so" : "Cannot load library /app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04/platforms/libqeglfs.so: (libQt5EglFSDeviceIntegration.so.5: cannot open shared object file: No such file or directory)"
qt.qpa.plugin: Could not load the Qt platform plugin "eglfs" in "/app/CoppeliaSim_Edu_V4_0_0_Ubuntu16_04" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted (core dumped)
```
Any help?

valayDave, 2020-03-25:
Hi @bycn . I have created a public docker image `valaygaurang/pyrep`

Can you try it out. It has copellia sim and Pyrep. 

Try running : 
```
xhost +local:docker
docker run -i -t --rm -e DISPLAY=$DISPLAY -u root -v /tmp/.X11-unix:/tmp/.X11-unix:ro --name="PyRep_Docker_Test" valaygaurang/pyrep:0.1
```

I can have tested this on docker in ubuntu1804.

bycn, 2020-03-25:
Hi @valayDave,
Thanks for your reply. To give more context, it's on a headless machine (no display)

bycn, 2020-03-25:
I ran your docker anyways locally and it seems to use xcb plugin. Could you please try to add `export QT_QPA_PLATFORM=eglfs`
?

valayDave, 2020-03-25:
@bycn : The docker Image I have provided run using CMD. U can try : 

```Dockerfile
FROM valaygaurang/pyrep:0.1

ENV QT_QPA_PLATFORM=eglfs
```

stepjam, 2020-03-25:
Here's my Dockerfile for working with RLBench if that helps:

```
FROM tensorflow/tensorflow:1.15.2-gpu-py3

LABEL maintainer="slj12@ic.ac.uk"

WORKDIR "/my_container"

ENV ZSH_THEME agnoster

RUN apt-get update && apt-get install -y \
  zsh \
  wget \
  libglib2.0-0  \
  libgl1-mesa-glx \
  xcb \
  "^libxcb.*" \
  libx11-xcb-dev \
  libglu1-mesa-dev \
  libxrender-dev \
  libxi6 \
  libdbus-1-3 \
  libfontconfig1 \
  xvfb \
  tar \
  git \
  qtbase5-dev qtdeclarative5-dev libqt5webkit5-dev libsqlite3-dev qt5-default qttools5-dev-tools \
  && rm -rf /var/lib/apt/lists/*

# Install ohmyzsh
RUN sh -c "$(wget -O- https://raw.githubusercontent.com/ohmyzsh/ohmyzsh/master/tools/install.sh)"
#SHELL ["/bin/zsh", "-c"]

ENV COPPELIASIM_ROOT /my_container/Apps/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04
ENV LD_LIBRARY_PATH $LD_LIBRARY_PATH:$COPPELIASIM_ROOT:$COPPELIASIM_ROOT/platforms
ENV QT_QPA_PLATFORM_PLUGIN_PATH $COPPELIASIM_ROOT

############################################
# CoppeliaSim/PyRep/RLBench
############################################

RUN mkdir Apps && cd Apps && \
  wget https://www.coppeliarobotics.com/files/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04.tar.xz && \
  tar -xf CoppeliaSim_Edu_V4_0_0_Ubuntu18_04.tar.xz && \

  git clone https://github.com/stepjam/PyRep.git && \
  cd PyRep && \
  pip3 install -r requirements.txt && \
  python3 setup.py install --user && \
  cd .. && \

  git clone https://github.com/stepjam/RLBench.git && \
  cd RLBench && \
  git checkout 1.0.6 && \
  pip3 install -r requirements.txt && \
  python3 setup.py install --user && \
  cd ..

############################################
# Prepare dev env
############################################

RUN pip3 install --user tensorflow-probability==0.8 psutil

ADD entrypoint.sh /bin/entrypoint.sh
RUN chmod +x /bin/entrypoint.sh
ENTRYPOINT ["/bin/entrypoint.sh"]
```

Where entrypoint.sh is:
```
#!/bin/sh
Xvfb :99 -screen 0 1024x768x24 +extension GLX +render -noreset &
export DISPLAY=:99
exec "$@"
```

bycn, 2020-03-25:
@stepjam thank you! Trying this out now, however this looks like it's using xcb as well, do you have a tested setup using egl?

valayDave, 2020-03-25:
Thank you for this @stepjam . 

@stepjam: 

I was investigating the `Task`object to understand where the rewards come from. In the process I realized that we need to shape and set the rewards ourselves for the simulation and basically set our own rewards. 

Can you direct me to some examples I can see reward shaping examples for RLBench ?

bycn, 2020-03-25:
I’m interested in this as well, but I believe a simple way other libraries do it outside of simulation is to calculate reward using the returned low dim state

stepjam, 2020-03-26:
No I have not tested with egl.

Regarding the rewards: not quite; you don't set the rewards yourself (RLBench handles that for you), instead each task registers 'success conditions'. When all of the success conditions are met, then a reward of 1 is returned. RLBench tries to encourage this sparse reward. For example, in the [reach target task](https://github.com/stepjam/RLBench/blob/master/rlbench/tasks/reach_target.py#L19), we register 1 condition that the tip of the robot should be inside a given proximity sensor.

bycn, 2020-03-26:
Can you support egl? I can’t use hardware acceleration otherwise.

stepjam, 2020-03-27:
I can certainly look into this once I find some time. Can't give an ETA though, so happy to accept a pull request if you figure something out before me.

closed 2020-07-04 by stale[bot]

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless