stepjam/PyRep #153: How to avoid a runtime error for PyRep

Gloriabhsfer, 2020-03-25:
Hi Stephen:
Thanks for the awsome package. I am using the package to train a reinforcement learning model with UR5 in Vrep. But when I update my go position for reinforcement learning ( I use DQN in this case), it cannot let me go to the target position for each step.
My plan is to update my end effector through each step for one episode. But it will break the simulation and return my some error like this:

```
> ---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-5-0a3f86081f40> in <module>
      9         #env.render()
     10         action = RL.choose_action(observation)
---> 11         observation_, reward, done, info = env.step(action)
     12         ax,ay,az,tx,ty,tz = observation_
     13         #print(observation_)

<ipython-input-2-29980caa3d15> in step(self, action)
     84         robot_done=False
     85         while not robot_done:
---> 86             robot_done=self.path.step()
     87             self.pr.step()
     88             self.path.set_to_start()

~/.local/lib/python3.6/site-packages/pyrep/robots/configuration_paths/arm_configuration_path.py in step(self)
     49         if self._rml_handle is None:
     50             self._rml_handle = self._get_rml_handle()
---> 51         done = self._step_motion() == 1
     52         self._path_done = done
     53         return done

~/.local/lib/python3.6/site-packages/pyrep/robots/configuration_paths/arm_configuration_path.py in _step_motion(self)
    167         dt = sim.simGetSimulationTimeStep()
    168         lengths = self._get_path_point_lengths()
--> 169         state, posVelAccel = sim.simRMLStep(self._rml_handle, dt, 1)
    170         if state >= 0:
    171             pos = posVelAccel[0]

~/.local/lib/python3.6/site-packages/pyrep/backend/sim.py in simRMLStep(handle, timeStep, dofs)
    893     # timeStep = ffi.cast('double', timeStep)
    894     state = lib.simRMLStep(handle, timeStep, newPosVelAccel, ffi.NULL, ffi.NULL)
--> 895     _check_return(state)
    896     return state, list(newPosVelAccel)
    897 

~/.local/lib/python3.6/site-packages/pyrep/backend/sim.py in _check_return(ret)
     26     if ret < 0:
     27         raise RuntimeError(
---> 28             'The call failed on the V-REP side. Return value: %d' % ret)
     29 
     30 

RuntimeError: The call failed on the V-REP side. Return value: -1
```
If the controller cannot find a path, I make the episode end and let it go to another attemption. I am not very sure about your backend structure, could you tell me where I need to adjust in my low-level control so I can realize a single follow a point end-effector controller?

Best,
Yu Zhang

stepjam, 2020-03-26:
What control mode are you using? Are you using the `solve_ik` function or the `get_path` function to get to the target?

Also, I'm not sure what you are doing in line 3 here:
```
robot_done=self.path.step()
self.pr.step()
self.path.set_to_start()  # It's a little unusual to have this line here (in the loop)
```
Best,
Stephen

Gloriabhsfer, 2020-03-31:
Hi Stephen:
Sorry for the late update. My control mode is inverse kinematic. I use `get_path` inside the controller. My goal is to let the end-effector learning how to find a path to a random point inside the whole workspace.
I use bullet 2.78 engine.
Because I use DQN and it is discrete action space algorithm.My strategy is make the action space become 6 range for -x,+x,-y,+y,-z,+z and everystep I will get a action base the policy DQN gave me and let my end-effector move to this position base on IK. I firstly try the `solve_ik` but when I read the example for end-effectro control, it seems` get_path` can use some algorithm like SBL,RRT( which is inside the arm.py code) So I choose to use this function as my low level controller.
The line you quote is I need to decided the whether the robot is go to the target position or not. If it is done, I can calculate the reward function for this step and decided if I need more action to get the next step.
Best,
Gloria

closed 2020-07-04 by stale[bot]