stepjam/PyRep #154: how to record a video?

wawachen, 2020-03-25:
hi,
Is there any API we can use to record a video from each timestep?

stepjam, 2020-03-26:
There are many ways. One way with opencv is:

```
import numpy as np
from cv2 import VideoWriter, VideoWriter_fourcc

width = 1280
height = 720
fps = 24
fourcc = VideoWriter_fourcc(*'MP42')
video = VideoWriter('./my_vid.avi', fourcc, float(fps), (width, height))

pr = PyRep()
pr.launch(...)
cam = VisionSensor('my_cam')

for _ in range(100):
    img = (cam.capture_rgb() * 255).astype(np.uint8)
    video.write(img)
video.release()
```

Best,
Stephen

closed 2020-03-26 by stepjam

masud99r, 2020-06-13:
I think there is a minor bug in the above code sample. `capture_rgb()` returns in RGB format whereas `write()` expects BGR (reverse). 
A quick fix:
```
img = np.flip(img, 2) # reverse color properties
video.write(img)
```

Thanks for the sample, it works for me.
-Masud