stepjam/PyRep #155: Can not go to target quaternion or orientation in path planning

HeegerGao, 2020-03-27:
Hello, I am using PyRep for a Jaco arm planning task. I tried to set the quaternion or euler parameters in `arm.get_path()` function, but when I executed the path, the tip cannot end up with the target orientation as I set. It just ended up with another orientation which looks like a random orientation. The position of the tip is right.

I tried the example in this repo using your Panda arm, and I find there is no such kind of problems with Panda arm in path planning. I don't know if this issue comes from my jaco arm or something else(maybe the path planning function cannot find a proper path). Could you help me? Thank you!

stepjam, 2020-03-28:
It sounds like the Jaco 'tip' dummy is in the wrong pose by default. First check to see if the tip dummys are the same across Panda and Jaco?

HeegerGao, 2020-03-28:
> It sounds like the Jaco 'tip' dummy is in the wrong pose by default. First check to see if the tip dummys are the same across Panda and Jaco?

I am sorry that I am not quite sure about what you mean. Do you mean that the initial orientation of Jaco tip dummy and Panda tip dummy should be the same? They are indeed not the same, but they are both the same with their gripper respectively. I have changed the initial orientation of the tip dummy but the issue is still there. Actually I don't know why they should be the same orientation.

Here is my [scene file](https://cloud.tsinghua.edu.cn/f/7302cb51e5aa4556a0b3/) and a [very simple path planning code](https://cloud.tsinghua.edu.cn/f/eb9cd123887740668c64/). I hope you can try it to see the issue. Remember to change the file path in the code. You should see that every time the tip would end up with a different orientation although I have set it with a fixed orientation(`euler=[0, math.radians(180), 0]`). Thank you!

yzlc080733, 2020-03-31:
Hi.
I just found that the settings of the calculation module equipped with Jaco (in PyRep repository) is a bit different from that of other arms, e.g. Panda.
Please see "tools -- calculation modules -- edit IK elements -- Jaco tip". The "alpha-beta" and "gamma" are not turned on by default in my case. After selecting them I got the arm work as I expected.
Hope this may help.

closed 2020-07-04 by stale[bot]