stepjam/PyRep #156: path.step and pr.step is not synchronized

ancorasir, 2020-04-03:
I build a task similar to example_panda_reach_target.py.

and run the following code to execute the path
    done = False
    while not done:
        done = path.step()
        pr.step()

However, I found that the robot hasn't reached the target position in the GUI while path.step() returns true. 
If I execute pr.step() for a few times, the robot will finally reach the target position.

Could you suggest what might be the reason? Thank you.

My guess is set_simulation_timestep is too small so that pr.step() can not complete the update in path.step().

ancorasir, 2020-04-03:
Problem solved by uncheck the real simulation option in setting. Does any one know the reason?

stepjam, 2020-04-06:
Hi,
Was the 'real simulation option' selected by default, or did you originally select this?
Not sure exactly why this behaviour emerges, could be down to a simulation timestep mismatch.

closed 2020-07-04 by stale[bot]