stepjam/PyRep #158: Rendering of vision sensors on a headless machine

Ramomrito, 2020-04-06:
Hi,

I am currently trying to render vision sensor images on a remote cluster (without a display) using CoppeliaSim/PyRep, but I am facing an error regarding the offscreen rendering. I have tried running CoppeliaSim with xvfb-run, but I receive the following error:

`QXcbIntegration: Cannot create platform offscreen surface, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
`

I should also note that I was previously able to run CoppeliaSim/PyRep on the same cluster when my scenes did not include any vision sensors using xvfb-run. This issue seems to be directly related to the rendering of the vision sensors.

The error message seems to imply that a suitable 3D rendering extension is not enabled, but I am not entirely sure how to do so. Therefore, I was wondering if you have faced a similar issue and know of a way to perform offscreen rendering without the use of a display.

Please let me know if you need more information about this issue.

Thanks a lot,
Omar

stepjam, 2020-04-06:
Rather than doing fvfb-run, have you tired something like the following:

```
Xvfb :99 &
export DISPLAY=:99
python3 my_pyrep_prog.py
```

Ramomrito, 2020-04-06:
Hi Stephen,

I have tried your suggestion, but it does not seem to work on the cluster I am using. When I try to use Xvfb, I get an error about not being able to connect to a display:
`qt.qpa.xcb: could not connect to display lo-login-02:16.0`

If I use xvfb-run, however, the program can run long enough to get to the previous error I mentioned (neither GLX nor EGL are enabled).

To give you more context, the cluster that I am using is a load sharing facility (LSF) batch system, which means I need to submit computing jobs to run the program. After some investigation, I noticed that the login nodes of the cluster do have GLX enabled with the glxinfo command, but when I submit a job to execute glxinfo it crashes on the compute node also because it could not connect to a display:
`Error: unable to open display lo-login-02:16.0`

I am guessing this issue might be more related to this kind of cluster setup. I am not sure if you are familiar with this, but any help would be appreciated!

Omar

Ramomrito, 2020-04-06:
Hey,

Just a quick update:

When I export the display environment variable before executing Xvfb, I can avoid the `could not connect to display` error, but I then obtain the same old error:

```
QXcbIntegration: Cannot create platform offscreen surface, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
```

Another weird thing is that I can now successfully run glxinfo/glxgears on these compute nodes, when I wrap the commands with Xvfb like so:
```
#!/bin/sh
export DISPLAY=:99
Xvfb :99 -screen 0 1024x768x24
glxinfo | grep -i opengl
```
which should mean that GLX is available.

At this point I am confused as to why GLX is perceived as disabled when I run CoppeliaSim though.

mch5048, 2020-04-13:
@Ramomrito Hi, I'm worried that this question is not related to resolving the problem you're facing, but can I get some tips for using PyRep in the headless remote server? 
I'm trying to do so by first compiling CoppeliaSim in headless mode, and bind it with PyRep but, 
Errors discussed in #14 
I'd be grateful if you can give me some pointers on this problem, and I wish I could look into the 
problem you're facing with headless PyRep.
thanks!

Ramomrito, 2020-04-17:
Hi @mch5048 ,

I am not compiling CoppeliaSim from source in the pure headless mode (without Qt) because it does not provide any support for rendering images, which is what I need for my project.

Are you sure you need to compile CoppeliaSim without Qt to work on your headless server? Usually you can use something like xvfb-run, which can make the program run, but it will rely on software rendering. Can you try that?

Here are some CoppeliaSim forum posts that can help you:
[https://forum.coppeliarobotics.com/viewtopic.php?f=9&t=5095](url)
[https://forum.coppeliarobotics.com/viewtopic.php?t=8291](url)

Ramomrito, 2020-04-17:
Also, I finally figured out that in my case it was a configuration issue on the cluster. 

Since the cluster does not provide physical displays, the application had to be run with Xvfb, which only supports software rendering. The problem was that the libraries that perform software rendering (Mesa) were only partially installed on the cluster.

This cluster uses Centos 7 and the fix was to install the `mesa-dri-drivers.x86_64` package together with the regular mesa installation package `mesa-libGL`.

Closing this issue now, thanks for the help!

closed 2020-04-17 by Ramomrito

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless