stepjam/PyRep #161: Properly closing Copelia on a training scrip

nanbaima, 2020-04-10:
Hey Stepjam,

Quick question, I'm probably doing it wrong but, how do I properly finish/close the connection with the CopeliaSim trough a PyRep script?
Because every time I run my scrip, when it finishes, the Copelia is closed and all fine, but for the next time I want to run the same script on headless mode I have to open the CopeliaSim, click on ¨NO¨ at an box saying CopeliaSim Crash, and only than I can run the script again, otherwise it will be waiting and wont be able to train. If I run in normal mode, the same box appears I only after I click on ¨NO¨ it starts.
So I am basically closing/finishing my program the wrong way.

[image]

I am using [https://github.com/vitchyr/rlkit.git](url) for that:
`if __name__ == "__main__":
    # noinspection PyTypeChecker
    variant = dict(
        algorithm="SAC",
        version="normal",
        net_size=256,
        replay_buffer_size=int(1E6),
        algorithm_kwargs=dict(
            num_epochs=2500,
            num_eval_steps_per_epoch=500,
            num_trains_per_train_loop=100,
            num_expl_steps_per_train_loop=200,
            min_num_steps_before_training=100,
            max_path_length=200,
            batch_size=512,
        ),
        trainer_kwargs=dict(
            discount=0.99,
            soft_target_tau=5e-3,
            target_update_period=1,
            policy_lr=3E-4,
            qf_lr=3E-4,
            reward_scale=1,
            use_automatic_entropy_tuning=True,
        ),
    )
    setup_logger('name-of-experiment', variant=variant)
    #ptu.set_gpu_mode(False)  # optionally set the GPU (default=False)
    experiment(variant)`

ycps, 2020-05-06:
Hi Renan,
did you try adding a [close method](https://github.com/openai/gym/blob/2c53496e2c0aa4bc8abae5bd753a7bf30866651a/gym/core.py#L105) to your custom env that calls [shutdown](https://github.com/stepjam/PyRep/blob/70c760ccbad69296e62ab56b5b93f8c9a8a09b11/pyrep/pyrep.py#L140)?
Something like this:
```
import gym
import pyrep
#[…]
class MyCustomEnv(gym.Env):
    def __init__(self):
        self.pr = pyrep.PyRep()
#[…]
    def close(self):
        self.pr.shutdown()
```

nanbaima, 2020-05-07:
> Hi Renan,
> did you try adding a [close method](https://github.com/openai/gym/blob/2c53496e2c0aa4bc8abae5bd753a7bf30866651a/gym/core.py#L105) to your custom env that calls [...]

Hey Yuri,
Yeap, I did, does the method has to be named  > close?
I've created the method shutdown on my env:
`#[…]
    def shutdown(self):
        self.pr.stop()
        self.pr.shutdown()
`

nanbaima, 2020-05-16:
Yeah, just worked here.
Thanks

closed 2020-05-16 by nanbaima