stepjam/PyRep #164: Resetting environment problem and vision sensor image quality problem

wawachen, 2020-04-15:
Unlike other simulators,  I find your **reset settings** for the V-rep environment is not **complete** and may result in some problems. 

In your example code, the reset is only recovering the position or pose and Not consider the velocity (although the dynamics are reset). At the beginning of the episode, it still has some **drift**.

Another problem is that, if my robot is accidentally broken (some items are falling down) due to the crash or collision in one episode,  for the next episode, it is not wise for you to only reset the pose. The right way is that you have to remove the previous model and import a new robot.
------------------------------------------------------------------------------------------------------------
For a vision sensor, I use the vision sensor to capture the image by 
`img = (py_env.cam.capture_rgb() * 255).astype(np.uint8)`

however, when I save the picture, it looks like below,
[image]
The color is not good and appears to be more dark (not like the picture from the float viewing in the scene). My vision sensor setting is below,

[image]

can you give me some feedback?

stepjam, 2020-04-15:
Hi,

regarding the reset: the example I give is just a simple example, and so you can of course implement these types of resets in your project. 

regarding the image: how are you displaying the image? It perhaps looks like you are uisng OpenCV to display, and if that is the case then remember that OpenCV expects BGR rather than RGB.

wawachen, 2020-04-15:
Thanks, it works.

closed 2020-04-15 by wawachen