stepjam/PyRep #165: Reinforcement learning error

minkyo96, 2020-04-16:
Hi. I am a student doing research related to reinforcement learning.

I changed the agent code of reinforcement learning in the example to a Jaco bot(J2n6s300) and trained it with a PPO algorithm.

When the learning time became longer, the joint connection of the r

obot was disconnected.

In order to solve this phenomenon, I reset the environment by stopping and starting vrep every reset of the learning environment.

I decided that the cause of the phenomenon was in the set_joint_position function.

However, the set_joint_position function is needed for learning, so if the disconnection of the robot is another cause, I wonder what the cause is.

In the environment that I learned, I attach a video of the state of learning.

[image]

KONPEITO1205, 2020-04-20:
Hi, I have the same problem.
I avoided it by calling `pyrep.stop()` and `pyrep.start()`.
This makes restart the environment.
So I could use `set_joint_position()`.

I attach my environment code, you can refer to.
```
class ReacherEnv(object):

    def __init__(self):
        self.pr = PyRep()
        self.pr.launch(SCENE_FILE, headless=False)
        self.pr.start()
        self.agent = Dobot()
        self.agent.set_control_loop_enabled(False)
        self.agent.set_motor_locked_at_zero_velocity(True)
        self.end_effector = Dummy('Dobot_tip')
        self.goal = Shape('goal')
        self.agent_ee_tip = self.agent.get_tip()
        self.distance_threshold = 0.05
        self.initial_joint_positions = self.agent.get_joint_positions()

    def restart(self):
        self.pr.stop()
        self.pr.start()
        self.agent = Dobot()
        self.agent.set_control_loop_enabled(False)
        self.agent.set_motor_locked_at_zero_velocity(True)
        self.end_effector = Dummy('Dobot_tip')
        self.goal = Shape('goal')
        self.agent_ee_tip = self.agent.get_tip()

    def _get_state(self):
        # Return state containing arm joint angles/velocities & target position
        return np.concatenate([self.agent.get_joint_positions(),
                               self.agent.get_joint_velocities()])

    def reset(self):
        self.restart()
        # Get a random position within a cuboid and set the target position
        goal_pos = list(np.random.uniform(POS_MIN, POS_MAX))
        self.goal.set_position(goal_pos)
        self.agent.set_joint_positions(self.initial_joint_positions)
        return self._get_state(), self.goal.get_position()
```

minkyo96, 2020-04-20:
Thank you for answer.

As you said, I solved this problem by "restarting" the pyrep.
In this case, controlling the robot arm using velocity control is not a problem.

However, I want to control the robot arm using position control.
For position control, set_joint_position () must be called for each type step.
So since I can't use "restart", I think we need to solve the fundamental problem of the set_join_position () function.

timtody, 2020-04-22:
Hello,
You don't need to restart pyrep. Just be sure to set the joint target velocities to 0 when resetting an episode. This is not done in the example RL script but definitely necessary to prevent joint drift.

timtody, 2020-04-22:
Modified method would look something like this

`def reset(self):
        # Get a random position within a cuboid and set the target position
        pos = list(np.random.uniform(POS_MIN, POS_MAX))
        self.target.set_position(pos)
        self.agent.set_joint_positions(self.initial_joint_positions)
        # add this
        self.agent.set_joint_target_velocities([0] * len(self.agent.joints))

        return self._get_state()`

closed 2020-07-04 by stale[bot]

Shaluols, 2020-09-17:
> Modified method would look something like this
>
> ```
> def reset(self):
>
>
>         # Get a random position within a cuboid and set the target position
>         pos = [...]

Hi, I tried your method but did not succeed. 
Here is my reset code:
```
    def reset(self):
        POS_MIN, POS_MAX = [0.8, -0.2, 1.0], [1.0, 0.2, 1.4]
        # Get a random position within a cuboid and set the target position
        self.arm.set_joint_mode(JointMode.FORCE)
        self.arm.set_motor_locked_at_zero_velocity(True)
        pos = list(np.random.uniform(POS_MIN, POS_MAX))
        self.target.set_position(pos)

        self.arm.set_joint_positions(self.initial_joint_positions)
        self.arm.set_joint_target_velocities([0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        print(self.arm.get_joint_velocities())
        self.pr.step()

        return self._get_state()
```
After setting the joint target velocities to zero, I retrieved the velocities is zero which is correct. 
However, after several minutes of RL training, the joints start to drift, as you can see in the image:
[image]

Do you know the reason? 
Thanks in advance~

njanirudh, 2021-02-23:
Were you able to solve this issue?

Shaluols, 2021-02-23:
I did not continue with this project, as I recall, I did not succeed with this method, but you can reset the scene to make the links on the correct positions.

jianye0428, 2021-05-13:
Hello,
 I worked also on a reinforcement learning problem with PyRep and struggle with the reset problem. With the example in the repository I added the line code `self.agent.set_joint_target_velocities([0] * len(self.agent.joints))` but it didn't work. 

I want to know if there is another solution to solve this problem?

thanks in advance.

Jian

njanirudh, 2021-05-15:
@jianye0428 
I was able to train the model successfully using coppeliaSim and stable-baselines3.
[RL project](https://github.com/NJ-2020-thesis/PyRep/blob/feature/examples/vmp/vmp_environment.py)

you can look at setup_scene() function.

# Try these and check if it solves your problem
self.agent.reset_dynamic_object()
self.pr.set_configuration_tree(self.agent_state)

jianye0428, 2021-05-16:
@njanirudh 

It worked！！Thank you so much！！

lgzid, 2023-08-12:
@jianye0428 I have discovered a new problem. The joint angle has been successfully reset in the above situation, but the joint speed has not been reset. Does this affect the results of reinforcement learning? If so, how can we solve the problem of joint speed

daylight1128, 2025-07-08:
@lgzid 
Have you resolved the issue of the speed not being reset? Does it affect reinforcement learning? After resetting the speed, executing pr.step() multiple times can reduce it to 0, but I'm not sure if this is reasonable
`for _ in range(15):
            self.pr.step()`