stepjam/PyRep #166: Camera stops streaming on changing camera resolution

Harsharma2308, 2020-04-20:
I want to change the resolution of a camera.
On changing the camera resolution by calling
`cam.set_capture_resolution((size,size))`
The camera just stops streaming images.
How do I set the camera resolution to be higher?

wawachen, 2020-04-20:
Hi,
I haven’t tried set_capture_resolution()function
But My solution is that
1. Directly Change the resolution parameters of your camera in the scene first.
2. Back to the code, call the cam.capture_rgb() then the output image will match the resolution you set in the scene.

To check the current resolution of your camera, you can call 
cam.get_resolution() 
Result is [width, height]

Harsharma2308, 2020-04-21:
Sorry updated `set_capture_resolution` to `set_resolution`. Every time I set the resolution, that particular vision sensor gets paused I guess, and calling capture_rgb shows a complete black image. Other cameras do show the images on calling capture_rgb. 
I am also using rlbench. It starts working when I run some other command in the environment. So I cannot say if it is a PyRep issue entirely. Closing this for now.

closed 2020-04-21 by Harsharma2308