stepjam/PyRep #168: PyRep in Google Colab

ycps, 2020-05-07:
Hello, I have trying to run CoppeliaSim+PyRep on Google Colab, but ran into issues.
The full ipynb is reproduced below:
```
!wget 'https://coppeliarobotics.com/files/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04.tar.xz'
!tar -xf 'CoppeliaSim_Edu_V4_0_0_Ubuntu18_04.tar.xz'

import os
COPPELIASIM_ROOT = '/content/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04'
os.environ['COPPELIASIM_ROOT'] = COPPELIASIM_ROOT
os.environ['LD_LIBRARY_PATH'] += f':{COPPELIASIM_ROOT}'
os.environ['QT_QPA_PLATFORM_PLUGIN_PATH'] = COPPELIASIM_ROOT

!git clone https://github.com/stepjam/PyRep.git
%cd PyRep
%pip install -r requirements.txt
!python setup.py install --user
%cd /content

import pyrep # triggers error
```
error:
```
/root/.local/lib/python3.6/site-packages/pyrep/backend/sim.py in <module>()
      1 from .simConst import *
----> 2 from ._sim_cffi import ffi, lib
      3 import numpy as np
      4 import collections
      5 

ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory
```
Which is odd, because we can confirm that the file/link exists:
```
os.path.exists(os.path.join(os.environ['COPPELIASIM_ROOT'], 'libcoppeliaSim.so.1'))
# True
```

nanbaima, 2020-05-16:
Good question!

Skylion007, 2020-06-01:
Setting the LD_LIBRARY_PATH inside of collab won't affect the currently running python process, only newly spawned python and other subprocesses. (Doing ```!python -c 'import pyrep'``` should load the library correctly). You can solve that issue at least by doing a more aggressive install and copy the shared libraries directly into the currently loaded library path like so:
```python3
from distutils.dir_util import copy_tree
copy_tree(COPPELIASIM_ROOT, '/usr/lib/')
```

closed 2020-08-07 by stale[bot]