stepjam/PyRep #174: Headless mode doubt - Error: signal 11:

nanbaima, 2020-05-21:
Hey Stephen,
How are you?

I have a question, I hope you can help me with that:
What should be the difference between running headless using a virtual framebuffer such as xvfb, and not using it but just setting pr.launch(SCENE_FILE, headless=True)?
I'm running on my env ` xvfb-run python3 my_pyrep_app.py` and not setting the env to headless True, so I'm setting it like `pr.launch(SCENE_FILE, headless=False)`. It works fine, however it seems that it is not running faster than the regular mode.

I've tried to set the env to headless True and not using the xvfb, in other words: `pr.launch(SCENE_FILE, headless=True)`, ` python3 my_pyrep_app.py` .  and it runs faster but few steps after the first simulation, the env just crashes and I get the Erro signal 11.

I've also tried to set the env to headless True using the xvfb, in other words: `pr.launch(SCENE_FILE, headless=True)`, `xvfb-run python3 my_pyrep_app.py`.  And it also runs faster but again, with few steps after the first simulation, the env also crashes and it gives the Erro signal 11.
 
I saw your answer to an old post that someone was getting a similar Error Code. [Error Signal 11 #23](https://github.com/stepjam/PyRep/issues/23#issuecomment-512272652). As you can see [here](https://github.com/stepjam/PyRep/issues/161#issue-598061361), I was indeed making a similar mistake, but now I'm using the close method to my custom env. However since I started to work with PyRep, I was running my code in headless False, with the xvfb. Now I few that it is not good enought.

Any ideas or suggestions?

`Error: signal 11:

/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_Z11_segHandleri+0x2b)[0x7f1f68ea10bb]
/lib/x86_64-linux-gnu/libc.so.6(+0x3ef20)[0x7f1f8836df20]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5Gui.so.5(_ZNK14QOpenGLContext16qGLContextHandleEv+0x4)[0x7f1f677ae0c4]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5OpenGL.so.5(_ZN10QGLContext17fromOpenGLContextEP14QOpenGLContext+0x13)[0x7f1f686db2a3]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5OpenGL.so.5(+0x221a2)[0x7f1f686db1a2]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5OpenGL.so.5(_ZN10QGLContextC2EP14QOpenGLContext+0xad)[0x7f1f686db27d]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5OpenGL.so.5(_ZN10QGLContext17fromOpenGLContextEP14QOpenGLContext+0x48)[0x7f1f686db2d8]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5OpenGL.so.5(+0x221a2)[0x7f1f686db1a2]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5OpenGL.so.5(_ZN10QGLContextC2EP14QOpenGLContext+0xad)[0x7f1f686db27d]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libQt5OpenGL.so.5(_ZN10QGLContext17fromOpenGLContextEP14QOpenGLContext+0x48)[0x7f1f686db2d8]
QMutex: destroying locked mutex`

nanbaima, 2020-06-11:
Apparently I found out that starting the physics of the Coppelia (`pyrep.start()`) on the init of the environment is not a good idea.
I moved to do it inside the `reset()` method, and when I run headless mode by Pyrep it works now.
(i.e. `pr.launch(SCENE_FILE, headless=True)`, `python3 my_pyrep_app.py`)
However there is still one problem regarding` xvfb-run` and running headlessly with it, I'm still getting the message `Erro 11` when using it.
(i.e. `pr.launch(SCENE_FILE, headless=True)`, `xvfb-run python3 my_pyrep_app.py`)

closed 2020-08-17 by stale[bot]

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless