stepjam/PyRep #178: Unable to allocate section memory!

nanbaima, 2020-06-08:
Hey Stephan,
How are you!?

I'm using PyRep with a Reinforcement Learning libs since a while now, it keeps controlling the interaction and training using PyRep, during my training after a lot of interactions with PyRep, something like 1500 times, it seems that PyRep is allocating a lot of memory, and when that happens I get Error messages such as:

```
LLVM ERROR: Unable to allocate section memory! 
QMutex: destroying locked mutex
```

Do you have any suggestions, to keep that from happening?
Or idea of what I migth be doing worng?

nanbaima, 2020-07-14:
Now getting errors such as:
```
Error: signal 11:

/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_Z11_segHandleri+0x2b)[0x7fb99a4530bb]
/lib/x86_64-linux-gnu/libc.so.6(+0x3efd0)[0x7fb9b996bfd0]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x5faf0f)[0x7fb8d14b6f0f]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x5fd668)[0x7fb8d14b9668]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x72a95b)[0x7fb8d15e695b]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0xdadd5)[0x7fb8d0f96dd5]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0xdde92)[0x7fb8d0f99e92]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN13CVisionSensor18renderForDetectionEibbbbRKSt6vectorIiSaIiEE+0xef)[0x7fb99a30e90f]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN13CVisionSensor13detectEntity2Eibbbb+0xb7)[0x7fb99a3121d7]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN13CVisionSensor44detectVisionSensorEntity_executedViaUiThreadEibbbb+0x75)[0x7fb99a312505]
QMutex: destroying locked mutex
```
Is it the same mistake or not?

nanbaima, 2020-07-22:
Update of same Error 11 stop, with different Hex position ID:

```
Error: signal 11:

/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_Z11_segHandleri+0x2b)[0x7f4a2e93e0bb]
/lib/x86_64-linux-gnu/libc.so.6(+0x3efd0)[0x7f4a4de56fd0]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x5faf0f)[0x7f496e082f0f]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x5fd668)[0x7f496e085668]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x72a95b)[0x7f496e1b295b]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0xdadd5)[0x7f496db62dd5]
/usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0xdde92)[0x7f496db65e92]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN13CVisionSensor18renderForDetectionEibbbbRKSt6vectorIiSaIiEE+0xef)[0x7f4a2e7f990f]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN13CVisionSensor13detectEntity2Eibbbb+0xb7)[0x7f4a2e7fd1d7]
/home/nanbaima/Software/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so.1(_ZN13CVisionSensor44detectVisionSensorEntity_executedViaUiThreadEibbbb+0x75)[0x7f4a2e7fd505]
QMutex: destroying locked mutex

```

closed 2020-09-27 by stale[bot]

ummagumm-a, 2024-04-13:
Hi @nanbaima. Did you manage to solve this issue?