stepjam/PyRep #181: when restart environment output version informations

takeru1205, 2020-06-13:
Hi, there

I'm using PyRep for reinforcement learning environment.
When I reset the environments, once stop PyRep and start again.
And PyRep displays some version information.
I want not to display them.
Is there a way not to display them?

nanbaima, 2020-06-13:
Hey @takeru1205,
Take a look at the Pyrep launch doc, at [pyrep.launch](https://pyrep.readthedocs.io/en/latest/pyrep.html?highlight=launch#pyrep.pyrep.PyRep.launch) there are the `write_coppeliasim_stdout_to_file` parameter, simply set it to `True` when launching the Pyrep and all output data will be stored in a /tmp file.

takeru1205, 2020-06-14:
Thanks, @nanbaima,
Amazing! You are a genius!!
I am really grateful to you!!

closed 2020-06-14 by takeru1205

nanbaima, 2020-06-14:
Hey @takeru1205 hahaha 
Thanks dude,
But actually, you have to thanks this guy over here: #173 
He is the one who in fact did something really nice, I only pointed it out to you.

fuying-wang, 2020-08-20:
Hi @nanbaima, I am wondering if I can just suppress the log of Coppeliasim instead of all output data, because I also want to print something to check the progress of my program. Thanks!