stepjam/PyRep #183: solve_ik error with my added robot

takeru1205, 2020-06-23:
Hi there,

I added Dobot Magician environment.
I want to use solve_ik method but it can get an error.

```
from os.path import dirname, join, abspath
from pyrep import PyRep
from pyrep.robots.arms.dobot import Dobot

SCENE_FILE = join(dirname(abspath(__file__)), 'scene_dobot_camera.ttt')

DELTA = 0.01
pr = PyRep()
pr.launch(SCENE_FILE, headless=False)
pr.start()
agent = Dobot()
starting_joint_positions = agent.get_joint_positions()
pos, quat = agent.get_tip().get_position(), agent.get_tip().get_quaternion()

def move(index, delta):
    pos[index] += delta
    new_joint_angles = agent.solve_ik(pos, quaternion=quat)
    agent.set_joint_target_positions(new_joint_angles)
    pr.step()

[move(2, -DELTA) for _ in range(20)]
[move(1, -DELTA) for _ in range(20)]
[move(2, DELTA) for _ in range(10)]
[move(1, DELTA) for _ in range(20)]

pr.stop()
pr.shutdown()

```

And I got error below.
```
Traceback (most recent call last):
  File "dobot_ik.py", line 23, in <module>
    [move(2, -DELTA) for _ in range(20)]
  File "dobot_ik.py", line 23, in <listcomp>
    [move(2, -DELTA) for _ in range(20)]
  File "dobot_ik.py", line 18, in move
    new_joint_angles = agent.solve_ik(pos, quaternion=quat)
  File "/home/takeru/.local/lib/python3.7/site-packages/pyrep/robots/arms/arm.py", line 140, in solve_ik
    raise IKError('IK failed. Perhaps the distance was between the tip '
pyrep.errors.IKError: IK failed. Perhaps the distance was between the tip  and target was too large.

```

And this is [my scene_file](https://drive.google.com/file/d/1ecXjZ3TsFMrPcEoPkiQ0MMQFWwoVALFs/view?usp=sharing).

The code is refer to [this panda example code](https://github.com/stepjam/PyRep/blob/master/examples/example_panda_end_effector_control.py). I only changed to my dobot scene.

If you have any ideas, please help me.

Thank you.

closed 2020-08-29 by stale[bot]

njanirudh, 2020-09-15:
Hi were you able to solve this issue

takeru1205, 2020-09-17:
Unfortunately, I couldn't solve this issue.