stepjam/PyRep #187: Consistency with real Franka Panda robot

ancorasir, 2020-06-28:
Hi James,

Thanks for the contribution. Recently I have been using both simulated panda in pyrep and real robot. I found that with the same joint configuration, the pose of the end-effector is not consistent between simulation and real robot.

My home joint configuration is home_joints = [0, -np.pi/4, 0, -3 * np.pi/4, 0, np.pi/2, np.pi/4]

pose of end-effector for real robot:
array([[ 9.99990e-01,  4.12553e-04, -3.61046e-04,  3.06873e-01],
       [ 4.12456e-04, -9.99990e-01, -2.69668e-04, -4.57676e-05],
       [-3.61154e-04,  2.69516e-04, -1.00000e+00,  4.86701e-01],
       [ 0.00000e+00,  0.00000e+00,  0.00000e+00,  1.00000e+00]])

pose of tip with reference to the robot base in simulation:
array([[ 9.97415185e-01, -7.31776936e-06,  7.18548745e-02, 3.61055901e-01],
       [-6.48048945e-06, -1.00000000e+00, -1.18855069e-05,  -8.16938002e-03],
       [ 7.18548745e-02,  1.13891301e-05, -9.97415185e-01, 5.18177509e-01],
       [ 0.00000000e+00,  0.00000000e+00,  0.00000000e+00, 1.00000000e+00]])

In simulation, I have add a base link at the bottom of the base link and the tip frame is rotated 180 degrees around the z axis so that these two frames in the simulation are consistent with the real robot. As you can see, the rotations of the tip are consistent in simulation and real robot but the positions are not. 
[image]

ancorasir, 2020-06-28:
The difference in the x-coordinate position can be due to the misplacement of the base frame in the x-axis and can be compensated by shifting the base frame along the x-axis. But the difference in the z-coordinate is difficult to explain as the base frame should be at the bottom of the base link as shown in the figure above.

hazimian, 2020-08-24:
Not sure if this is relevant to this issue, but I found a discrepancy between the PyRep Panda model and another model available here https://github.com/marcocognetti/FrankaEmikaPandaDynModel. If I remember correctly, joint 4 is off by a few degrees.

closed 2020-10-30 by stale[bot]