stepjam/PyRep #189: Problem when installing - missing coppeliasim.lib

ansonlei, 2020-07-02:
I am trying to install PyRep on windows 10 with CoppeliaSimEdu and I received this error:

LINK : fatal error LNK1181: cannot open input file 'coppeliaSim.lib'
[image]

I realised that there is no coppeliaSim.lib file in the installation directory of CoppeliaSim. I have tried to reinstall CoppeliaSimEdu but it's still missing.

nanbaima, 2020-07-02:
PyRep is not supported on Windows yet. [see the end of this topic](https://github.com/stepjam/PyRep#install)

> Although you can use CoppeliaSim on any platform, communication via PyRep is currently only supported on Linux.

However, you might take a look at this #169 commit, try to check if it works for you, comment, and try to find the best way to implement it and help others.
I hope it helps you!

ansonlei, 2020-07-02:
It works well. Thanks for your help.

closed 2020-08-02 by stepjam