stepjam/PyRep #191: ImportError: libicui18n.so.56: cannot open shared object file: No such file or directory

fuying-wang, 2020-07-03:
I am trying to use PyRep in ubuntu 18.04. It seems the installation works well. But when i ran the exmaple code in `examples`, i received this error:
```
python example_turtlebot_navigation.py
Traceback (most recent call last):
  File "example_turtlebot_navigation.py", line 7, in <module>
    from pyrep import PyRep
  File "/home/fuying/.local/lib/python3.7/site-packages/pyrep/__init__.py", line 6, in <module>
    from .pyrep import PyRep
  File "/home/fuying/.local/lib/python3.7/site-packages/pyrep/pyrep.py", line 3, in <module>
    from pyrep.backend import sim, utils
  File "/home/fuying/.local/lib/python3.7/site-packages/pyrep/backend/sim.py", line 2, in <module>
    from ._sim_cffi import ffi, lib
ImportError: libicui18n.so.56: cannot open shared object file: No such file or directory
```
I tried to google it and it seems to be the version problem of `icu` package. But I still can not figure it out. Could you help me about this?

stepjam, 2020-07-06:
Not seen this issue before. After doing some google, it sounds like you may have 2 versions of QT installed. Maybe check for that first?

fuying-wang, 2020-07-08:
Thanks for your reply!

I solved this problem by reinstalling `CoppeliaSim`.

closed 2020-07-08 by fuying-wang

nanbaima, 2020-07-08:
> Thanks for your reply!
> 
> I solved this problem by reinstalling `CoppeliaSim`.

Hey @fuying-wang could you please share how did you exactly reinstall? (the method, how, just download? or any specificity)
Because I keep getting these errors #174 and It looks something like what you presented here.

fuying-wang, 2020-07-17:
Hi,

Sorry for the late reply. My problem is some links in `CoppeliaSim ` are invalid. So I deleted the previous 'CoppeliaSim' folder and redownloaded it from [this link](https://www.coppeliarobotics.com/downloads.html). Then it works.