stepjam/PyRep #195: supporting VRep

hazimian, 2020-07-09:
Hi There
I am wondering if the latest version of PyRep still supports VRep. If so what are the installation steps?

Thanks

H.A.

nanbaima, 2020-07-14:
Hey Hazimian,
Not sure if this information helps you:
[V-Rep/Coppelia Support](https://github.com/stepjam/PyRep#what-happened-to-v-rep)

stepjam, 2020-08-02:
if you __really__  need to use V-REP, the I would recommend to use the [last commit before PyRep converted to CoppeliaSim](https://github.com/stepjam/PyRep/tree/dc255b52bfd3819149d7541be2de583173fc4973), as @nanbaima pointed out. 

However, I don't see any reason why you would need top use V-REP, CoppeliaSim is just a renaming/rebranding of V-REP.

closed 2020-08-02 by stepjam