stepjam/PyRep #198: ModuleNotFoundError: No module named 'pyrep'

ranjn2, 2020-07-15:
I was trying to run the examples given in the repository from PyRep/examples after following the installation steps.

I am running into the following issue when running an example script from either  /examples or from outside the root project directory:

python example_panda_reach_target.py 

Traceback (most recent call last):
  File "example_panda_reach_target.py", line 8, in <module>
    from pyrep import PyRep
ModuleNotFoundError: No module named 'pyrep'

My python version is 3.7. Did anyone else face this when trying to run any of the example programs, does anyone have any 
Thanks!

nanbaima, 2020-07-16:
Hey @ranjn2,
Just to be sure, did you check if you installed with (pip) python2 ou (pip3) python3?
In other world is PyRep being properly installed in your 3.7 Python package file system?
Or even maybe you might be running with a different user, so you might not have access to the user's installed package file. 
Just giving a few ideas that you may check.

stepjam, 2020-08-02:
@ranjn2 Sounds like it was not installed in the right place. Can you make sure pyrep is shown when you run pip3; e.g. `pip3 list | grep PyRep`. May also be worth checking that your default python version is python3, so try explicitly running: `python3 example_panda_reach_target.py`

closed 2020-08-02 by stepjam

amaciRL2021, 2021-03-23:
Hi @stepjam what did you mean when you say "right place"? What is the right place? I have the same problem and if I try to show pyrep with pip3 list | grep PyRep, it does not appear. Thank you in advance!

Yasmin-Hesham, 2021-05-29:
While following your steps in installation, in line "pip3 install .", I got this error:

Defaulting to user installation because normal site-packages is not writeable
Processing /home/yassmin/PyRep
  DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at https://github.com/pypa/pip/issues/7555.
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-e4pu2bln/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-e4pu2bln/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-e3ev69gu                  
         cwd: /tmp/pip-req-build-e4pu2bln/                                             
    Complete output (7 lines):                                                         
    Traceback (most recent call last):                                                 
      File "<string>", line 1, in <module>                                             
      File "/tmp/pip-req-build-e4pu2bln/setup.py", line 7, in <module>                 
        import cffi_build.cffi_build as cffi_build                                     
      File "/tmp/pip-req-build-e4pu2bln/cffi_build/cffi_build.py", line 12, in <module>
        raise RuntimeError('COPPELIASIM_ROOT defined, but is not a valid path.')       
    RuntimeError: COPPELIASIM_ROOT defined, but is not a valid path.                   
    ----------------------------------------                                           
WARNING: Discarding file:///home/yassmin/PyRep. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

Knowing that CopeliaSim is installed successfully, cloned the repo, added the paths in bashrc file and sourced it. 
Could you please help?

hlhfhmt, 2021-09-17:
> While following your steps in installation, in line "pip3 install .", I got this error:
> 
> Defaulting to user installation because normal site-packages is not writeable
> Processing /home/yassmin/PyRep
> DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
> pip 21.3 will remove support for this functionality. You can find discussion regarding this at [pypa/pip#7555](https://github.com/pypa/pip/issues/7555).
> ERROR: Command errored out with exit status 1:
> command: /usr/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-e4pu2bln/setup.py'"'"'; **file**='"'"'/tmp/pip-req-build-e4pu2bln/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(**file**) if os.path.exists(**file**) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, **file**, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-e3ev69gu
> cwd: /tmp/pip-req-build-e4pu2bln/
> Complete output (7 lines):
> Traceback (most recent call last):
> File "", line 1, in
> File "/tmp/pip-req-build-e4pu2bln/setup.py", line 7, in
> import cffi_build.cffi_build as cffi_build
> File "/tmp/pip-req-build-e4pu2bln/cffi_build/cffi_build.py", line 12, in
> raise RuntimeError('COPPELIASIM_ROOT defined, but is not a valid path.')
> RuntimeError: COPPELIASIM_ROOT defined, but is not a valid path.
> ----------------------------------------
> WARNING: Discarding file:///home/yassmin/PyRep. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
> ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
> 
> Knowing that CopeliaSim is installed successfully, cloned the repo, added the paths in bashrc file and sourced it.
> Could you please help?

same problem with you> how you deal this?

moongerim, 2021-12-28:
Try to install CoppeliaSim 4.1 version

MohammadSahandi, 2024-01-28:
Hi, For the problem of " COPPELIASIM_ROOT defined, but is not a valid path" you should write the path of coppeliasim installation in `export COPPELIASIM_ROOT=EDIT/ME/PATH/TO/COPPELIASIM/INSTALL/DIR` in `~/.bashrc` file.

For example if you have installed coppeliasim in desktop you should write this in `~/.bashrc` file:
`export COPPELIASIM_ROOT=/home/your_dvice_name/Desktop/Coppeliasim_file_name`