stepjam/PyRep #199: Run threaded child script using PyRep.script_call()

DarkGeekMS, 2020-07-22:
Hello there,

I was trying to run a simple threaded child script function using PyRep.script_call(), however it failed.

Remote lua procedure :
```
function test_function(inInts, inFloats, inStrings, inBuffer)
    return {6}, {6.0}, {'test'}, 'test'
end
```

Python code:
```
pr = PyRep()
pr.launch("test_scene_pyrep.ttt", headless=True)
ret = pr.script_call("test_function@Dummy", vrepConst.sim_scripttype_childscript) 
print(ret)
```

 This returns a runtime error:
```
RuntimeError: The call failed on the V-REP side. Return value: -1
```

Meanwhile, the same function runs normally, when it's a customization script.

So, any help running threaded child scripts using PyRep?

Thanks in advance.

Coppelia, 2020-07-22:
Hello,
my best guess is that your threaded child script has not yet been initialized, or has already ended. Make sure this is not the case

DarkGeekMS, 2020-07-22:
Thanks a lot for your response.

That's correct, it was an initialization issue.

I added a dummy loop to `sysCall_threadmain`  to initialize the child script:
```
while sim.getSimulationState()~=sim.simulation_advancing_abouttostop do
    print('test')
end
```

and started the simulation and stepped one time in python code :
```
pr.start()
pr.step()
```

Then everything worked as expected.

Thanks a lot for your help!

closed 2020-07-22 by DarkGeekMS