stepjam/PyRep #202: Failed to run exmaple

wkoa, 2020-08-11:
In my case,

- Python 3.6

- CoppelSim4.0.0

- ubuntu18.04

When I run `python3 example_baxter_pick_and_pass.py ` in the dir `PyRep/examples`, I got

_python3: Relink `/lib/x86_64-linux-gnu/libsystemd.so.0' with `/lib/x86_64-linux-gnu/librt.so.1' for IFUNC symbol `clock_gettime'
python3: Relink `/lib/x86_64-linux-gnu/libudev.so.1' with `/lib/x86_64-linux-gnu/librt.so.1' for IFUNC symbol `clock_gettime'
Segmentation fault (core dumped)_

stepjam, 2020-09-18:
Hi. Are you able to run coppeliaSim without PyRep?

wkoa, 2020-11-18:
> Hi. Are you able to run coppeliaSim without PyRep?

Yes, of course

stepjam, 2020-11-18:
OK let's start by reinstalling the latest version of CoppeliaSim and PyRep; just to make sure there's no version mismatch, and then we can go from there. They should be both at version 4.1

closed 2021-01-24 by stale[bot]