stepjam/PyRep #203: Macos Support

trifasico, 2020-08-16:
Hi, sorry for leaving this comentary here, but its the only way i have ton ask this.
Im a portuguese student and im experimenting and trying Coppelia sim and coding robots in a simualtion
But a i own a mac, i understand that for me to code i have de b0-remoteAPI, but i have read that this one is a lot faster, can you guys tell me when your going to release PyRep support for macos? or even windows?

I can help if you want!
Thanks, Best regards

nanbaima, 2020-08-17:
Oi Macos,
Tudo bem?
Então cara, quanto ao suporte ao MacOS a galera não tem falado nada de problema ou etc, até por se basear em UNIX tbm então não sei se teria muito pq ter problema, o que eu tenho visto é mais umas adaptações para rodar no windows pela problematica de endereçamento e tal, ai da uma olhada no pull do #169.
Espero ter ajudado :)

tingelst, 2020-09-01:
I have done some initial tests for MacOS support. Unfortunately, this will most likely not be possible due to how GUI threads are handled on MacOS. That is, you are only allowed to do GUI operations from the main thread, which is not how PyRep is currently implemented.

trifasico, 2020-09-10:
Isn’t it possible to do with no gui?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-11-17 by stale[bot]