stepjam/PyRep #204: Unable to use GPU for rendering

prasoongoyal, 2020-08-29:
I can run an example script using xvfb: `xvfb-run python examples/example_baxter_pick_and_pass.py`, but this doesn't use the GPU for rendering (as expected).

But, when I try running it without xvfb -- `python examples/example_baxter_pick_and_pass.py`, I get the following error:
```
[CoppeliaSim:loadinfo]   CoppeliaSim V4.1.0., (rev. 1), flavor: 1
[CoppeliaSim:loadinfo]   Legacy machine ID: 5000-9DEB-FFC4-9CEC-F7E3-841D
[CoppeliaSim:loadinfo]   Machine ID: C28C-736A-596C-0000-167E-0101
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/path/to/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04/" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

Aborted (core dumped)
```
Not sure how to fix this.

komms, 2020-09-02:
It could be the display problem. Assign right display to the user might solve.

prasoongoyal, 2020-09-03:
Thanks for the response! To clarify, does this need an X server, or can I use the GPU for rendering without X as well (I'm on a server that doesn't allow running X).

komms, 2020-09-04:
afaik Xorg is needed and you need to configure it. I'm not sure about whether one should use GPU or CPU on the server.

prasoongoyal, 2020-09-04:
Thanks. I guess leave this open for a bit to see if someone else might be able to add something about using GPU for rendering without X.

closed 2020-09-14 by prasoongoyal

kamaleshrathinam, 2021-01-21:
I have the same issue, trying to run Pyrep on Google Cloud for faster Rendering

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless