stepjam/PyRep #207: wrong quaternion order

lianghongzhuo, 2020-09-18:
I tried to use the [code](https://github.com/stepjam/PyRep/blob/master/pyrep/objects/object.py#L200-L223)
In the docstring it says quaternion (x,y,z,w).
However, I found that the correct quaternion order is Q(z, y, x, w).

I got this result by set a PrimitiveShape orientation using `set_orientation` using Eula angle and then using `get_quaternion` to got the quaternion value that Coppelia thinks is.

I suggest either update the documentation, or I can help change the order of the input quaternion to the correct order.

stepjam, 2020-09-18:
Hi. The quaternion function/doc-string is correct. Mixing euler angles and quaternions is usually never a good idea.
Please see here for more information regarding CoppeliaSims quaternion function: https://www.coppeliarobotics.com/helpFiles/en/regularApi/simSetObjectQuaternion.htm

closed 2020-09-18 by stepjam

lianghongzhuo, 2020-09-18:
Interesting, but it still did not solve my problem. Either the Coppelia is doing something wrong or the transform is wrong. Here is a sample code to reproduce my issues:
```
    from pyquaternion import Quaternion
    from pyrep import PyRep
    from pyrep.const import PrimitiveShape

    pr = PyRep()
    pr.launch("some/scene/file.ttt"), headless=False)
    pr.start()
    shape1 = Shape.create(type=PrimitiveShape.CUBOID, size=[0.2, 0.2, 0.2], color=(1, 0, 0), static=True,
                          position=[0.5, 0, 0.9])
    shape2 = Shape.create(type=PrimitiveShape.CUBOID, size=[0.2, 0.2, 0.2], color=(0, 1, 0), static=True,
                          position=[0.8, 0, 0.9])
    ori = Quaternion(axis=[1, 0, 0], angle=np.pi/4)
    shape1.set_orientation(ori.yaw_pitch_roll)
    shape2.set_quaternion([ori.x, ori.y, ori.z, ori.w])
    pr.step()
    print(shape1.get_quaternion())
    print(shape2.get_quaternion())
```
The output of the above code is:
`array([0.        , 0.        , 0.38268346, 0.9238795 ])`
and
`array([0.38268343, 0.        , 0.        , 0.92387956])`
But if the transform from the quaternion to the Eula angle is correct, the output should be the same.
Could you give me some suggestions on how this happened?
Thank you!

lianghongzhuo, 2020-09-21:
my fault, we need to use roll pitch yaw instead of yaw pitch roll.
so the quaternion order in doc is correct. Thank you.