stepjam/PyRep #210: Get stuck when running example.

newbieyxy, 2020-09-22:
Hello Stephen!
When I run an example, the program get stuck after printing the following information:
`[CoppeliaSim:loadinfo]   CoppeliaSim V4.1.0., (rev. 1), flavor: 1`
`[CoppeliaSim:loadinfo]   Legacy machine ID: 5000-9DEB-FFC4-9CEC-F7E3-841D`
`[CoppeliaSim:loadinfo]   Machine ID: 2B8B-736C-465E-0000-B15B-0101`

I think maybe this problem happens due to the calling of function `lib.simExtLaunchUIThread`, as the function `sim.simExtCanInitSimThread()` always returns False when launching scene in object `PyRep`, but I don't know how to fix it.

I have obsessed about this problem for a while, and I will be very grateful if you can provide some help!

stepjam, 2020-11-18:
Hi, 
Is this still an issue?
If so, does this happen when you launch CoppeliaSim without PyRep?

closed 2021-01-24 by stale[bot]

MajesticKhan, 2021-09-01:
Hi,

I had the same issue when training an agent using stablebaseline. The tricky part is evaluation during training as it needs a separate thread to run another coppeliasim simulator.

I tried running the training with evaluation using the example as stated from [here](https://github.com/stepjam/PyRep/issues/157) but I got stuck at 

```
        while not sim.simExtCanInitSimThread():
            time.sleep(0.1)
```

Looking at the documentation from multiprocessing library for [python 3.8](https://docs.python.org/3.8/library/multiprocessing.html) this seems to work much better:

```
import multiprocessing as mp

def foo(q):
    q.put('hello')

if __name__ == '__main__':
    ctx = mp.get_context('spawn')
    q = ctx.Queue()
    p = ctx.Process(target=foo, args=(q,))
    p.start()
    print(q.get())
    p.join()
```

It seems that one has to explicitly mention to start a brand new python process through `ctx = mp.get_context('spawn')`.

Hope this helps!