stepjam/PyRep #222: Example program hangs in headless mode

lianwenzhao, 2020-11-07:
Hi, when I ran an example in headless mode with xvfb, I got the following execution and it hangs there after `Reached target 5!`.
```
$ xvfb-run python ~/repos/PyRep/examples/example_panda_reach_target.py 
Reached target 0!
Reached target 1!
Reached target 2!
Reached target 3!
Reached target 4!
Reached target 5!
^C (SIGINT)
The X11 connection broke (error 1). Did the X11 server die?
QMutex: destroying locked mutex
QMutex: destroying locked mutex
```
But the expected behavior with GUI should be:
```
$ python ~/repos/PyRep/examples/example_panda_reach_target.py 
Reached target 0!
Reached target 1!
Reached target 2!
Reached target 3!
Reached target 4!
Reached target 5!
Reached target 6!
Reached target 7!
Reached target 8!
Reached target 9!
[CoppeliaSim:loadinfo]   done.
QObject::~QObject: Timers cannot be stopped from another thread
```
Do you have any suggestions? Thanks!

stepjam, 2020-11-18:
Hmm, that is strange... 
What happens if you instead run the following:
```
Xvfb :99 -screen 0 1024x768x24 +extension GLX +render -noreset &
export DISPLAY=:99
python ~/repos/PyRep/examples/example_panda_reach_target.py 
```

lianwenzhao, 2020-11-23:
Thanks! Didn't have bandwidth to try recently. Temporarily closing.

stepjam, 2021-02-03:
Reopening due to: https://github.com/stepjam/RLBench/issues/104

stepjam, 2021-02-03:
@lianwenzhao, did you ever resolve this issue?

stepjam, 2021-02-03:
I think this is an issue regarding the X11 server hanging, but not sure what exactly is causing it. We could try and alternative to Xvfb?

shagunsodhani, 2021-02-03:
What would be steps for using with xvfb?

stepjam, 2021-02-03:
The steps for Xvfb? You mentioned you were already running with Xvfb on the RLBench issue, right?

shagunsodhani, 2021-02-03:
Ohh my bad. I misinterpreted your comment "We could try and alternative to Xvfb?" to mean that I should switch to xvfb. You are right, I am already on xvfb.

stepjam, 2021-02-03:
It sounds like an issue with Xvfb rather than CoppeliaSim to be honest.
Maybe try running glxgears for a few minutes and see if that too hangs?
What platform are you running on by the way?

shagunsodhani, 2021-02-03:
1. `Maybe try running glxgears for a few minutes and see if that too hangs?` - sorry not familiar with `glxgears`.
2. I am on ubuntu 20.04 Python 3.8.6

stepjam, 2021-02-03:
`sudo apt install mesa-utils`

then:

`glxgears`

Making sure to have your X11 server running, e.g.

```
Xvfb :99 -screen 0 1024x768x24 +extension GLX +render -noreset &
export DISPLAY=:99
glxgears
```

shagunsodhani, 2021-02-03:
Seems to be working good.

```
(rlbench) ➜  examples git:(master) glxgears
9754 frames in 5.0 seconds = 1950.793 FPS
10064 frames in 5.0 seconds = 2012.788 FPS
9994 frames in 5.0 seconds = 1998.774 FPS
10032 frames in 5.0 seconds = 2006.364 FPS
9980 frames in 5.0 seconds = 1995.981 FPS
9990 frames in 5.0 seconds = 1997.901 FPS
10066 frames in 5.0 seconds = 2013.010 FPS
10066 frames in 5.0 seconds = 2013.199 FPS
9905 frames in 5.0 seconds = 1980.915 FPS
10125 frames in 5.0 seconds = 2024.927 FPS
10108 frames in 5.0 seconds = 2021.408 FPS
9848 frames in 5.0 seconds = 1969.383 FPS
```

shagunsodhani, 2021-02-03:
Confirming that `glxgears` works without any problem (have been running it for a few hours now)

lianwenzhao, 2021-02-04:
> @lianwenzhao, did you ever resolve this issue?
Didn't really go back to that. But the issue seems resolved from the conversation.

shagunsodhani, 2021-02-15:
@stepjam checking if there is anything else I could try.

closed 2021-04-24 by stale[bot]

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless

cshizhe, 2022-03-29:
I met the same issue. The problem is that I used a mismatched coppelia version with my system. The problem is solved by using the correct coppelia version.