stepjam/PyRep #224: ImportError: /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5: version `Qt_5' not found

thanard, 2020-11-17:
I had followed the suggestion in #76, and encounter the following error when trying to import PyRep.

```
Traceback (most recent call last): File "<stdin>", line 1, in <module> 
File "/home/thanard/anaconda3/lib/python3.7/site-packages/pyrep/__init__.py", line 6, in <module> 
   from .pyrep import PyRep 
File "/home/thanard/anaconda3/lib/python3.7/site-packages/pyrep/pyrep.py", line 3, in <module> 
   from pyrep.backend import sim, utils 
File "/home/thanard/anaconda3/lib/python3.7/site-packages/pyrep/backend/sim.py", line 2, in <module> 
   from ._sim_cffi import ffi, lib ImportError: /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5: version Qt_5 not found (required by /home/thanard/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu16_04/libcoppeliaSim.so.1)
```

I also checked and found that `/usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5` exists.

lianghongzhuo, 2020-11-17:
Remove all the qt related package in your conda environment should fix this.

thanard, 2020-11-17:
@lianghongzhuo  Thanks for your quick response! I followed the steps here:

1. conda list | grep 'qt'
2. Check what packages do you have with qt
3. conda remove --force <INSERT_EACH_PACKAGE> (--force only removes the package and not its dependencies)
4. try to run it again

Then, I ran into the problem above.

lianghongzhuo, 2020-11-17:
What is your OS version?

thanard, 2020-11-17:
> What is your OS version?

16.04

lianghongzhuo, 2020-11-17:
What's your terminal output if you run `conda list |grep qt`?

stepjam, 2020-11-17:
Have you tried reordering the path variables? (#94)
I mostly use virtualenvs and docker, so I've never come across these anaconda problems.
However, a large number of people have had these anaconda problems with various fixes, so it would be nice to compile a list of anaconda _gotchas_ and add them to the README.

lianghongzhuo, 2020-11-17:
I could not reproduce your issue in a ubuntu16.04 computer. Could you try to install pyrep in a newly created conda environment?
For example run `conda create -n new_env`

thanard, 2020-11-17:
conda list |grep qt returns nothing.

I tried creating a new environment and still ran into the same issue. Even without conda it also encounters this issue, so it might not be a conda related issue.

stepjam, 2020-11-17:
> I had followed the suggestion in #76, and encounter the following error when trying to import PyRep.
>
> ```
> Traceback (most recent call last): File "<stdin>", line 1, in <module>
> File [...]

Just to confirm, you get this during or after install? I thought the issue was installing, but it looks like this is an error when running pyrep?

stepjam, 2020-11-17:
Oh sorry, I misread your initial issue. I thought this was regarding installation. Woops!
Can you make sire you are not in the project root when importing pyrep. I think that should solve your issue.

thanard, 2020-11-17:
Thank you @stepjam and @lianghongzhuo. Yes it was when loading pyrep, and I made sure that I called out outside the repo. Also, I tried installing qt5-default and qttools5-dev-tools. However, the same issue still persists.

stepjam, 2020-11-17:
Hmm strange.
Whats the output of running:
```echo $LD_LIBRARY_PATH // $COPPELIASIM_ROOT // $QT_QPA_PLATFORM_PLUGIN_PATH```

thanard, 2020-11-17:
/home/thanard/tamp_work/gurobi901/linux64/lib:/home/thanard/.mujoco/mujoco200/bin:/home/thanard/.mujoco/mujoco200/bin:/usr/lib/nvidia-450:/usr/local/cuda/lib64/:/usr/lib/x86_64-linux-gnu/:/home/thanard/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu16_04 // /home/thanard/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu16_04 // /home/thanard/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu16_04

thanard, 2020-11-17:
Thanks @stepjam. It works now! Essentially there is qt5 in both `/usr/lib/x86_64-linux-gnu/` and `/home/thanard/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu16_04`. So removing the first one from LD_LIBRARY_PATH fixed the issue.

closed 2020-11-17 by thanard