stepjam/PyRep #225: How to do headless rendering?

thanard, 2020-11-17:
Is there an example script for headless rendering?

stepjam, 2020-11-18:
[It's one of the launch options](https://github.com/stepjam/PyRep/blob/master/pyrep/pyrep.py#L99)
```
pr = PyRep()
pr.launch(headless=True)
```
If the machine you are running on **does not have a display attached**, then you will additionally need to launch your program using [Xvfb](https://github.com/stepjam/PyRep#running-headless).

closed 2020-11-18 by stepjam

thanard, 2020-11-18:
Actually, what I mean is how do we get image outputs from pr.

stepjam, 2020-11-18:
You'll want to use the [VisionSensor](https://github.com/stepjam/PyRep/blob/master/pyrep/objects/vision_sensor.py) class:
```
pr = PyRep()
pr.launch('my_scene.ttt', headless=True)
vs = VisionSensor('my_sensor')
rgb = vs.capture_rgb()
depth = vs.capture_depth()
```
Where 'my_sensor' is the name of the vision sensor as defined in the scene.

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless