stepjam/PyRep #228: running headless error

xxx-007, 2020-11-21:
hello ,after installing pyrep,i run headless,my code is :
`sudo apt-get install xvfb `
`xvfb-run python3 examples/example_baxter_pick_and_pass.py`

but i get an error :
`ImportError: /home/xiximu/anaconda3/envs/pyrep_env/lib/python3.6/site-packages/pyrep/backend/_sim_cffi.cpython-36m-x86_64-linux-gnu.so: undefined symbol: simAddLog`

i do not know what i should do then i can solve this problem .thank you !

lianghongzhuo, 2020-11-21:
Did you set this line: https://github.com/stepjam/PyRep/blob/master/examples/example_baxter_pick_and_pass.py#L18
to 
`pr.launch(SCENE_FILE, headless=Trur)`?

xxx-007, 2020-11-21:
no ,i do not , l only run `xvfb-run python3 examples/example_baxter_pick_and_pass.py` in terminal, then i meet the error.

and i also try `pr.launch(SCENE_FILE, headless=True)`,but when i run at `from pyrep import PyRep`,i meet other error :`ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory`

lianghongzhuo, 2020-11-21:
headless mode means you do not have a gui interface. simulation run in terminal. so you have to set `headless = True`

xxx-007, 2020-11-21:
i set `headless=True`,but i still have the error :`ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory`,my code can not run at `from pyrep import PyRep`,there is  a image:
[image]

lianghongzhuo, 2020-11-21:
Could you also try:
```
cd example
python3 example_baxter_pick_and_pass.py
```

xxx-007, 2020-11-21:
that also cannot solve my problem.now the condition is this :
if i run in terminal ,there is a error:
[image]

if i run in pytharm ,there is another error:
[image]

i do not know how to solve those two problems.

lianghongzhuo, 2020-11-21:
Is your Coppelia has the newest version? The current version is 4.1

stepjam, 2020-11-21:
This to me looks like a PyRep-CoppeliaSim version mismatch. What version of PyRep and CoppeliaSim are you using?
Also, can you run the following:
```
echo $LD_LIBRARY_PATH // $COPPELIASIM_ROOT // $QT_QPA_PLATFORM_PLUGIN_PATH
```

xxx-007, 2020-11-21:
vrep is CoppeliaSim_Edu_V4_0_0_Ubuntu16_04
[image]

PyRep is version='4.1.0'

stepjam, 2020-11-21:
Please update your CoppeliaSim

xxx-007, 2020-11-21:
after updating,vrep is CoppeliaSim_Edu_V4_1_0_Ubuntu16_04
[image]

my code is 
[image]

but there still  have the error
[image]

stepjam, 2020-11-21:
What happens when you run from terminal?
```
cd example
python3 example_baxter_pick_and_pass.py
```

xxx-007, 2020-11-21:
in terminal ,i can correctly run and i do not meet any errors.but in pycharm,i still have the error.

stepjam, 2020-11-21:
You are getting the error in pycharm because you are running the program from the pyrep root. When running from the pyrep root, it will instead use the local uninstalled version, which is not what you want. See: https://github.com/stepjam/PyRep#troubleshooting)

closed 2020-11-21 by stepjam

taitoushiyun, 2021-01-09:
> in terminal ,i can correctly run and i do not meet any errors.but in pycharm,i still have the error.

> You are getting the error in pycharm because you are running the program from the pyrep root. When running from the pyrep root, it will instead use the local uninstalled version, which is not what you want. See: https://github.com/stepjam/PyRep#troubleshooting)

i meet the same problem while running the program outside the pyrep root. the true reason is that in terminal we have set the environment variables of LD_LIBRARY_PATH, COPPELIASIM_ROOT and QT_QPA_PLATFORM_PLUGIN_PATH in ~/.bashrc, but in Pycharm which is not set by default. by setting these environment variables in Pycharm>RUN>Edit Configurations>Environment variables, the problem has been fixed.

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless