stepjam/PyRep #247: Installation with python 3.8 fails

orensbruli, 2021-01-05:
With this error when compiling cffi
```
    c/call_python.c:20:30: error: dereferencing pointer to incomplete type ‘PyInterpreterState’ {aka ‘struct _is’}
       20 |     builtins = tstate->interp->builtins;
          |                              ^~
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```
Is the requirement to cffi==1.11.5 fixed to that version for some reason?
Thank you in advance.

zfwang615, 2021-01-06:
Hi, I met the same error almost the same time.
I have cheked and cffi == 1.14.4 is installed with no error.
So I just changed cffi version to "cffi==1.14.4" in the 2nd line of requirement.txt and completed the installation.
Thank the great work!^_^

zfwang615, 2021-01-06:
> With this error when compiling cffi
> 
> ```
>     c/call_python.c:20:30: error: dereferencing pointer to incomplete type ‘PyInterpreterState’ {aka ‘struct _is’}
>        20 |     builtins = tstate->interp->builtins;
>           |                              ^~
>     error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
> ```
> 
> Is the requirement to cffi==1.11.5 fixed to that version for some reason?
> I have checked and cffi==1.12.3 is installed with no error.
> Thank you in advance.

Hi, I met the same error almost the same time.
I have cheked and cffi == 1.14.4 is installed with no error.
So I just changed cffi version to "cffi==1.14.4" in the 2nd line of requirement.txt and completed the installation.
Thank the great work!^_^

zfwang615, 2021-01-06:
Hi，I met the same error almost at the same time yesterday.
I have checked and cffi==1.14.2 is installed with no error.
I just changed the cffi version to "cffi==1.14.2"（2nd line of requirement.txt）and completed the installation.
Thank the great work.^_^

orensbruli, 2021-01-07:
@zfwang615  and have you been able to use PyRep lib after compiling?

orensbruli, 2021-01-07:
With the chenge in requirements I get PyRep installed
[image]
but then I get this problem when trying to import:
[image]

lianghongzhuo, 2021-01-07:
@orensbruli Please copy and paste the text output instead of screenshotting. Screenshots cannot be searched nor copied from to reproduce.
For example:
```
your terminal output
```

zfwang615, 2021-01-08:
> @zfwang615 and have you been able to use PyRep lib after compiling?

Follow the readme.md to complete the installation.
`
pip install -r requirements
pip install .
`
After that, I just run the scripts in examples.
If you have some import error, maybe you can try to install some basic 'xxx-dev' packages.
By the way, the text is better than screenshotting, like @lianghongzhuo's advice.

orensbruli, 2021-01-08:
> Follow the readme.md to complete the installation.
> ` pip install -r requirements pip install .`

I did.

> After that, I just run the scripts in examples.
> If you have some import error, maybe you can try to install some basic 'xxx-dev' packages.

I still have the same problem. I'm gonna try in a clean VM. I guess it's currently some problem with my installation. 
Thank you for your help.

> By the way, the text is better than screenshotting, like @lianghongzhuo's advice.  

+1 Edited previous response. Thanks.

stepjam, 2021-01-08:
@orensbruli the error:
`
ImportError: /usr/local/lib/python3.8/dist-packages/pyrep/backend/_sim_cffi.cpython-38-x86_64-linux-gnu.so: undefined symbol: simAddLog
`

Looks like a CoppeliaSIm-PyRep version mismatch. The PyRep version corresponds to the CoppeliaSim version you should be using (in this case, 4.1.0).

orensbruli, 2021-01-08:
> @orensbruli the error:
> `ImportError: /usr/local/lib/python3.8/dist-packages/pyrep/backend/_sim_cffi.cpython-38-x86_64-linux-gnu.so: undefined symbol: simAddLog`
>
> Looks like a CoppeliaSIm-PyRep [...]
That is! Thank you so much.

orensbruli, 2021-01-08:
@stepjam what do you think about updating the requirents.txt file with a newer version of cffi?

closed 2021-01-08 by orensbruli