stepjam/PyRep #252: Error - undefined symbol: simAddLog

guilhermecano, 2021-01-11:
Good afternoon!
Yesterday I tried to instal the most recent version of PyRep, but it is not working. It seems to have a problem with simAddLog in both CoppeliaSim 4.0.0 and CopeliaSim 4.1.0.
The package version and the error that appears when I am trying to import PyRep are presented below:

`Name: PyRep
Version: 4.1.0`

`Python 3.7.9 (default, Aug 18 2020, 06:22:45) 
[GCC 7.5.0] on linux`

```
>>> from pyrep import pyrep
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/gui/Dsc/tools/PyRep/pyrep/__init__.py", line 6, in <module>
    from .pyrep import PyRep
  File "/home/gui/Dsc/tools/PyRep/pyrep/pyrep.py", line 3, in <module>
    from pyrep.backend import sim, utils
  File "/home/gui/Dsc/tools/PyRep/pyrep/backend/sim.py", line 2, in <module>
    from ._sim_cffi import ffi, lib
ImportError: /home/gui/Dsc/tools/PyRep/pyrep/backend/_sim_cffi.cpython-37m-x86_64-linux-gnu.so: undefined symbol: simAddLog
```

Does anyone know why this error occurs?
Thank you!

stepjam, 2021-01-11:
Looks like a PyRep - CoppeliaSim version mismatch. The version numbers should match up.
Can you make sure that when you tried CopeliaSim 4.1.0., that the LD_LIBRARY_PATH was correct?

guilhermecano, 2021-01-11:
I just verified again and yes, it is pointing to CopeliaSim 4.1.0 (rev 0 - Ubuntu 18).
```
export COPPELIASIM_ROOT=/home/gui/Dsc/tools/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$COPPELIASIM_ROOT
export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT
```
I double-checked with `cd $COPPELIASIM_ROOT` and I also echoed the LD_LIBRARY_PATH, it seems ok. If it is not a common problem among PyRep users, there may be something wrong here. I will keep on trying to run it and maybe try it on another machine with Ubuntu 20.

nanbaima, 2021-01-11:
Hey @guilhermecano,

I think I had this problem, and it was with sim_cffi I'm not sure now, but I've installed it on hard code and I think it solves this problem.
Maybe @refalguera had a similar problem as well.

stepjam, 2021-01-11:
OK, and have you tried reinstalling PyRep (rerunning `pip install .`) ?

refalguera, 2021-01-11:
Hey @guilhermecano,

Try using python 3.7.7, with this version of python my cffi related problems have stopped. If it doesn't work, try installing PyRep using the old installation version: python setup.py install, it worked for me.

guilhermecano, 2021-01-11:
Yes, I tried with both `pip install -e .` and `python setup.py install`, but I still got the same error. I guess will try it on another PC to ensure there is nothing wrong with my python version, my pc, or the way I installed it.

guilhermecano, 2021-01-11:
@refalguera 
Thanks. I will try it :)

closed 2021-01-20 by stepjam