stepjam/PyRep #255: ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory

AmrapaliKhandare, 2021-01-25:
I am getting this error while running example codes. I am running it using Python 3. and I am using CoppeliaSimEdu version V4.1.0
```
(rl) amrapali@amrapali:~/PyRep/examples$ python3 example_turtlebot_navigation.py 
Traceback (most recent call last):
  File "example_turtlebot_navigation.py", line 7, in <module>
    from pyrep import PyRep
  File "/home/amrapali/.local/lib/python3.6/site-packages/pyrep/__init__.py", line 6, in <module>
    from .pyrep import PyRep
  File "/home/amrapali/.local/lib/python3.6/site-packages/pyrep/pyrep.py", line 3, in <module>
    from pyrep.backend import sim, utils
  File "/home/amrapali/.local/lib/python3.6/site-packages/pyrep/backend/sim.py", line 2, in <module>
    from ._sim_cffi import ffi, lib
ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory
 ```

stepjam, 2021-01-25:
Hi. Are you using anaconda, or are you using virtual envs?
If you are using anaconda, then there are some known issues with this; see #93 #94.

muttequrashi, 2021-01-26:
I am having the same issue and I am not using anaconda

muttequrashi, 2021-01-26:
[image]

stepjam, 2021-01-26:
Whats the output of running:
`echo $LD_LIBRARY_PATH // $COPPELIASIM_ROOT // $QT_QPA_PLATFORM_PLUGIN_PATH
`

AmrapaliKhandare, 2021-01-27:
I am getting this output.
```
/opt/ros/melodic/lib:/home/amrapali/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04 // /home/amrapali/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04 // /home/amrapali/Downloads/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04

```

muttequrashi, 2021-01-30:
> Whats the output of running:
> `echo $LD_LIBRARY_PATH // $COPPELIASIM_ROOT // $QT_QPA_PLATFORM_PLUGIN_PATH `

My Issue was resolved it was due to incorrect path

closed 2021-04-07 by stale[bot]

basab-gupta, 2023-09-08:
It's been a long time since this issue was opened, but I thought that I would still chime in. I also had the exact same issue when running CoppeliaSimEdu version V4.1.0 on Ubuntu 20.04 on WSL2. I managed to fix the issue.

I think the main problem is that the symbolic link to file `libcoppeliaSim.so.1` is missing in the installation directory for this version (and possibly other older versions too but I can't be too sure) of CoppeliaSim. I think the trick is to manually create that symbolic link yourself. Here is a quick summary of the steps for Ubuntu:

1. Navigate to the installation directory of CoppeliaSim
2. Create the symbolic link using: `ln -s libcoppeliaSim.so libcoppeliaSim.so.1`
3. Verify that the link was indeed created by typing `ls -l libcoppeliaSim.so.1`. Your output should look something like this `libcoppeliaSim.so.1 -> libcoppeliaSim.so` 
4. Set up the rest of your environment variables as described in the main installation guide.
5. Do not forget to source the script once you're done setting up your environment variables with ` source ~/.bashrc`
6. Navigate to the PyRep folder and activate your virtual environment if you are using one.
7. Then navigate to the examples folder and try to run it again. Hopefully you shouldn't see this issue anymore.

I hope this helps others too! It's such a simple fix, but it took me forever to try and find it. @stepjam Do you think it makes sense to add this to the official installation steps as well?

JXSalian, 2025-02-25:
how to solve this problem

rafayaamirgull, 2025-03-17:
> I am getting this error while running example codes. I am running it using Python 3. and I am using CoppeliaSimEdu version V4.1.0
>
> ```
> (rl) amrapali@amrapali:~/PyRep/examples$ python3 [...]

I had the same issue. I was using zsh, I shifted to bash and it worked.