stepjam/PyRep #256: CFFI error is persisting

anonrobotics, 2021-01-27:
Hi there, I posted a similar error a few days ago, and I believe my issue was marked as resolved and closed after a response said that this should no longer be an issue. However, I am still having problems with this package. I received the following error:
ERROR: Command errored out with exit status 1:
     command: /usr/local/opt/python@3.9/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/2x/hcdrc1p965j30z1dc6h3w84cbp1jq1/T/pip-req-build-mi4hj4l5/setup.py'"'"'; __file__='"'"'/private/var/folders/2x/hcdrc1p965j30z1dc6h3w84cbp1jq1/T/pip-req-build-mi4hj4l5/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/2x/hcdrc1p965j30z1dc6h3w84cbp1jq1/T/pip-pip-egg-info-8i5zfqvl
         cwd: /private/var/folders/2x/hcdrc1p965j30z1dc6h3w84cbp1jq1/T/pip-req-build-mi4hj4l5/
    Complete output (8 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/2x/hcdrc1p965j30z1dc6h3w84cbp1jq1/T/pip-req-build-mi4hj4l5/setup.py", line 2, in <module>
        import cffi_build.cffi_build as cffi_build
      File "/private/var/folders/2x/hcdrc1p965j30z1dc6h3w84cbp1jq1/T/pip-req-build-mi4hj4l5/cffi_build/cffi_build.py", line 747, in <module>
        os.symlink(path, path + '.1')
    FileExistsError: [Errno 17] File exists: '/Users/myname/Desktop/SS/PoseEstimation/Sim-to-real/CoppeliaSim_Edu_V4_1_0_Mac/libcoppeliaSim.so' -> '/Users/myName/Desktop/SS/PoseEstimation/Sim-to-real/CoppeliaSim_Edu_V4_1_0_Mac/libcoppeliaSim.so.1'
    creating symlink: /Users/myName/Desktop/SS/PoseEstimation/Sim-to-real/CoppeliaSim_Edu_V4_1_0_Mac/libcoppeliaSim.so.1 -> /Users/myName/Desktop/SS/PoseEstimation/Sim-to-real/CoppeliaSim_Edu_V4_1_0_Mac/libcoppeliaSim.so
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

Would really appreciate some more help!

anonrobotics, 2021-01-27:
I am encountering this when I run pip install .

stepjam, 2021-01-28:
MAC is not supported. From the README:

`Although you can use CoppeliaSim on any platform, communication via PyRep is currently only supported on Linux.`

Your previous issue said this was also not working on Linux? Making sure to install the latest PyRep version and the latest updates to requirements.txt

closed 2021-04-05 by stale[bot]