stepjam/PyRep #257: pyrep error, plz help  ----->  RuntimeError: Handle youBot_m_joint1 does not exist.

octorber123, 2021-01-28:
Hello PyRep team, 

i am trying to run a python script that uses a robot and i get this error:

Traceback (most recent call last):
  File "testsim.py", line 25, in <module>
    robot = HolonomicBase(0, 4, 0, 'youBot', 12, 6, 0.035)
  File "/usr/local/lib/python3.8/dist-packages/PyRep-4.1.0-py3.8-linux-x86_64.egg/pyrep/robots/mobiles/holonomic_base.py", line 34, in __init__
    super().__init__(count, num_wheels, name)
  File "/usr/local/lib/python3.8/dist-packages/PyRep-4.1.0-py3.8-linux-x86_64.egg/pyrep/robots/mobiles/mobile_base.py", line 27, in __init__
    super().__init__(count, name, joint_names)
  File "/usr/local/lib/python3.8/dist-packages/PyRep-4.1.0-py3.8-linux-x86_64.egg/pyrep/robots/robot_component.py", line 24, in __init__
    self.joints = [Joint(jname + suffix)
  File "/usr/local/lib/python3.8/dist-packages/PyRep-4.1.0-py3.8-linux-x86_64.egg/pyrep/robots/robot_component.py", line 24, in <listcomp>
    self.joints = [Joint(jname + suffix)
  File "/usr/local/lib/python3.8/dist-packages/PyRep-4.1.0-py3.8-linux-x86_64.egg/pyrep/objects/joint.py", line 16, in __init__
    super().__init__(name_or_handle)
  File "/usr/local/lib/python3.8/dist-packages/PyRep-4.1.0-py3.8-linux-x86_64.egg/pyrep/objects/object.py", line 24, in __init__
    self._handle = sim.simGetObjectHandle(name_or_handle)
  File "/usr/local/lib/python3.8/dist-packages/PyRep-4.1.0-py3.8-linux-x86_64.egg/pyrep/backend/sim.py", line 94, in simGetObjectHandle
    raise RuntimeError('Handle %s does not exist.' % objectName)
RuntimeError: Handle youBot_m_joint1 does not exist.
QObject::~QObject: Timers cannot be stopped from another thread
QMutex: destroying locked mutex

when i comment out the robot code it works perfectly, can you please help me with this issue?

Thank you
octorber123

closed 2021-04-23 by stale[bot]

frietz58, 2021-04-28:
Just quickly putting this here: It appears the YouBot that we can drag into the scene from the hierarchy in CoppeliaSim differs from the one in the `examples/scene_youbot_navigation.ttt` file, see: [image]
The model with the name `youBot #0` is the one I dragged into the scene, which has different joint names.
I assume this is due to newer versions of CoppeliaSim, I encountered this issue with CoppeliaSim 4.1.0. Anyway, copying the robot from the example scene into your custom scene and using that model instead of the one from CoppeliaSim should prevent the runtime error.

nanbaima, 2021-04-28:
> Just quickly putting this here: It appears the YouBot that we can drag into the scene from the hierarchy in CoppeliaSim differs from the one in the `examples/scene_youbot_navigation.ttt` file, see: [...]

hey @frietz58 and @octorber123,

So the models you should use are the ones inside the [robots ttm files](https://github.com/stepjam/PyRep/tree/master/robot_ttms), those are the models that are exported after being pre-setted to properly work with PyRep hierarchical structure.
When someone whats to add a new robot they should follow the tutorial [here](https://github.com/stepjam/PyRep/blob/master/tutorials/adding_robots.md) in which the final model should be exported and storage on the file master/robot_ttm.

Hopefully, this helps and answers your questions!
By the way, this one is [Youbot](https://github.com/stepjam/PyRep/blob/master/robot_ttms/mobiles/youBot.ttm) model, located at [robot_ttms/mobiles](https://github.com/stepjam/PyRep/blob/master/robot_ttms/mobiles/).

frietz58, 2021-04-28:
Right, it even says so in the readme >.< 
Thanks for pointing that out, I had completely forgotten about it!