stepjam/PyRep #258: PyRep Qt libcoppeliasim.so.1 errors

jooker23, 2021-01-31:
I downloaded Pyrep as follows om my ubuntu 18.04. 
https://travis-ci.com/github/stepjam/PyRep.

When i used ‘’ python3 -m unittest discover tests’’ and  gym-vrep example, i got two different errors
1-

Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/home/tan/python-environments/env/bin/platforms" ...
loaded library "/home/tan/stepjam/PyRep/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04/platforms/libqxcb.so"
qt.qpa.xcb: could not connect to display :99.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/tan/stepjam/PyRep/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

Aborted (core dumped)
2-
xvfb-run python3 example_turtlebot_navigation.py
Traceback (most recent call last):
  File "example_turtlebot_navigation.py", line 7, in <module>
    from pyrep import PyRep
  File "/home/tan/python-environments/env/lib/python3.6/site-packages/pyrep/__init__.py", line 6, in <module>
    from .pyrep import PyRep
  File "/home/tan/python-environments/env/lib/python3.6/site-packages/pyrep/pyrep.py", line 3, in <module>
    from pyrep.backend import sim, utils
  File "/home/tan/python-environments/env/lib/python3.6/site-packages/pyrep/backend/sim.py", line 2, in <module>
    from ._sim_cffi import ffi, lib
ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory
 
My echo $LD_LIBRARY_PATH // $COPPELIASIM_ROOT // $QT_QPA_PLATFORM_PLUGIN_PATH  outputs;

:/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04:/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04/platforms:/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04:/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04/platforms:/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04:/CoppeliaSim_Edu_V4_1_0_Ubuntu18_04 // /CoppeliaSim_Edu_V4_1_0_Ubuntu18_04 // /CoppeliaSim_Edu_V4_1_0_Ubuntu18_04

How can i fix it. Please help me about this situation.
Thanks,

houyanxu, 2021-02-18:
I have the same issue with you. And I deal with it by creating a new conda env for PyRep and make sure your CoppeliaSim_Edu version is 4.1, python = 3.7.7.

closed 2021-04-28 by stale[bot]

Zed-Wu, 2021-07-06:
@houyanxu @jooker23  Have you solved the problem 1? I have the same error. I run the examples on my GPU server but there is no monitor. Is there an option I can choose to close the display?