stepjam/PyRep #265: [Question] CoppeliaSim -h flag instead of xvfb?

chisarie, 2021-02-27:
Hi, thank you very much for this nice library!

I had a question about running in headless mode. 
Using xvfb works on my machine, but unfortunately it doesn't support GPU, which makes the simulation very slow.
On the other hand, I saw that CoppeliaSim itself has an headless mode, which can be used by running `./CoppeliaSim.sh -h` (see documentation [https://www.coppeliarobotics.com/helpFiles/en/commandLine.htm](https://www.coppeliarobotics.com/helpFiles/en/commandLine.htm)). I tested this mode and it runs on my remote machine even without xvfb.

I was wondering if there is a way to run the CoppeliaSim headless mode through PyRep, hence avoiding the need of xvfb and being able to use gpu acceleration.
Thank you very much in advance, 
Eugenio

MRzNone, 2021-04-30:
I am running into the same issue.  Unable to use GPU makes the simulation very slow and un-usable.

stepjam, 2021-07-01:
You can now do GPU rendering: https://github.com/stepjam/PyRep/blob/master/README.md#running-headless

closed 2021-07-01 by stepjam