stepjam/PyRep #266: Error running PyRep on Ubuntu 20.04

Souphis, 2021-03-23:
Hi,

I've switch from Ubuntu 18.04 to Ubuntu 20.04 and i'm struggling with running my code. I installed probably all required packages (qt5, xcb etc.). Even though, when I'm trying to run my code this error shows:

> QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
> 
> 
> Error: signal 11:
> 
> /opt/CoppeliaSim/libcoppeliaSim.so.1(_Z11_segHandleri+0x30)[0x7faaa3764c90]
> /lib/x86_64-linux-gnu/libc.so.6(+0x46210)[0x7fab273bf210]
> /lib/x86_64-linux-gnu/libc.so.6(+0x18b675)[0x7fab27504675]
> /opt/CoppeliaSim/libcoppeliaSim.so.1(_Z15initGl_ifNeededv+0x11c)[0x7faaa390ae5c]
> /opt/CoppeliaSim/libcoppeliaSim.so.1(_ZN11CMainWindowC2Ev+0x61d)[0x7faaa3be3abd]
> /opt/CoppeliaSim/libcoppeliaSim.so.1(_ZN3App16createMainWindowEv+0x71)[0x7faaa38fb6e1]
> /opt/CoppeliaSim/libcoppeliaSim.so.1(_Z24simRunSimulator_internalPKciPFvvES2_S2_iS0_b+0x475)[0x7faaa377a195]
> /opt/CoppeliaSim/libcoppeliaSim.so.1(_Z29simExtLaunchUIThread_internalPKciS0_S0_+0x99)[0x7faaa377a439]
> /home/grzbae/.virtualenvs/gym/lib/python3.8/site-packages/pyrep/backend/_sim_cffi.cpython-38-x86_64-linux-gnu.so(+0x495d2)[0x7faaa41005d2]
> python3(PyCFunction_Call+0xfa)[0x5f305a]

Any ideas what's going on here?

realjsk, 2021-03-31:
I suffered from the same problem for a long time. This is what worked for me:

1. I installed miniconda to be able to manage different python environments easily (installing anaconda should be fine too).
2. With conda activated, I created a new environment which I dedicated for PyRep (I am also calling it PyRep): `conda create --name PyRep python=3` 
3. Activate the new environment: `conda activate PyRep` 
4. While in the PyRep environment, install PyRep (as described in https://github.com/stepjam/PyRep#install)
5. Define your QT_PLUGIN_PATH. In my case (Ubuntu 20.04), this is done by: `export QT_PLUGIN_PATH="/usr/lib/x86_64-linux-gnu/qt5/plugins"`
6. Now I can run the PyRep examples (always while the PyRep environment is activated).

You may wanna start with Step 5. May be that's all you need.
Good luck!

Souphis, 2021-03-31:
Well, I found a solution (in my case). In my project I'm using opencv-python. During setting up new env, pip donwloaded the latest version of opencv, the 4.5.*.*. Downgrading this package to 4.2.*.* fixed all problems.

closed 2021-04-17 by Souphis

LinxiFan, 2021-05-26:
The solution that works for me without downgrading packages is:
`sudo apt-get install -y qt5-default`

OlivierLessard, 2025-11-04:
I found that forcing coppeliaSim to use GLX with: 
```
unset QT_XCB_GL_INTEGRATION
export QT_QPA_PLATFORM=xcb
export DISPLAY=:0
export __GLX_VENDOR_LIBRARY_NAME=nvidia
```
fixed the issue for me