stepjam/PyRep #270: QObject::~QObject: Timers cannot be stopped from another thread

oridong, 2021-04-18:
ubuntu20.04
python 3.8.5
coppeliasim 4.1 edu
d@ubuntu:~/planning/Vrep/PyRep/examples$  cd /home/d/planning/Vrep/PyRep/examples ; /usr/bin/env /usr/bin/python3.8 /home/d/.vscode/extensions/ms-python.python-2021.3.680753044/pythonFiles/lib/python/debugpy/launcher 43561 -- /home/d/planning/Vrep/PyRep/examples/example_panda_reach_target.py 
Reached target 0!
Reached target 1!
Reached target 2!
Reached target 3!
Reached target 4!
Reached target 5!
Reached target 6!
Reached target 7!
Reached target 8!
Reached target 9!
[CoppeliaSim:loadinfo]   done.
QObject::~QObject: Timers cannot be stopped from another thread

Elissa0723, 2021-04-22:
Is this the right output of running this example?
I'm a little confusing about this. When I running the program, after finished running, it will show this and close coppelisSim, is this means operate well or fail?

closed 2021-06-28 by stale[bot]

Leilasjd, 2022-09-17:
Hi, did you notice what is the right output? how should solve this error "QObject: Timers cannot be stopped from another thread"?