stepjam/PyRep #273: libcoppeliaSim.so.1: cannot open shared object file in COLAB

alex-kozinov, 2021-05-05:
Hello!
I've tried to install PyRep in Google colab. But i had an issue with importing pyrep after installing. 
You can reproduce this error in
https://colab.research.google.com/drive/1vrpXEWFn-rI97dYwYNPAVYxwgfGPc36N?usp=sharing

lianghongzhuo, 2021-05-05:
I think pyrep can not use in colab. You need download CoppeliaSim and install pyrep locally.

alex-kozinov, 2021-05-06:
> I think pyrep can not use in colab. You need to download CoppeliaSim and install pyrep locally.

But why?  There is ubuntu 18.04 on virtual machine

closed 2021-07-12 by stale[bot]

scalemailted, 2023-03-17:
I also want to use PyRep in Google Colab. I installed it and reproduced the same error cited in this thread.

``` import pyrep```

> Traceback (most recent call last):
>  File "<stdin>", line 1, in <module>
>  File "/usr/local/lib/python3.9/dist-packages/pyrep/__init__.py", line 5, in <module>
>    from .pyrep import PyRep
>  File "/usr/local/lib/python3.9/dist-packages/pyrep/pyrep.py", line 3, in <module>
>    from pyrep.backend import sim, utils
>  File "/usr/local/lib/python3.9/dist-packages/pyrep/backend/sim.py", line 2, in <module>
>    from ._sim_cffi import ffi, lib
> ImportError: libcoppeliaSim.so: cannot open shared object file: No such file or directory

Is there a workaround for this issue? I am attempting to use the newest version: **4.4.0 rev 0**