stepjam/PyRep #275: ModuleNotFoundError: No module named 'pyrep.backend._sim_cffi'

Elsie7, 2021-05-16:
ModuleNotFoundError: No module named 'pyrep.backend._sim_cffi'

I have successfully installed Pyrep(with CoppeliaSim), it prompts an error like above when I try to import prep and start to write my first simulation script.
My script as below:
elsiechao@ubuntu:~/Documents/Tools/PyRep$ python3
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrep.pyrep
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/elsiechao/Documents/Tools/PyRep/pyrep/__init__.py", line 5, in <module>
    from .pyrep import PyRep
  File "/home/elsiechao/Documents/Tools/PyRep/pyrep/pyrep.py", line 3, in <module>
    from pyrep.backend import sim, utils
  File "/home/elsiechao/Documents/Tools/PyRep/pyrep/backend/sim.py", line 2, in <module>
    from ._sim_cffi import ffi, lib
ModuleNotFoundError: No module named 'pyrep.backend._sim_cffi'
>>> 
elsiechao@ubuntu:~/Documents/Tools/PyRep$ find |grep sim_cffi
elsiechao@ubuntu:~/Documents/Tools/PyRep$ find |grep simConst
./pyrep/backend/simConst.py
./pyrep/backend/__pycache__/simConst.cpython-38.pyc
./tools/generate_simConst.py
elsiechao@ubuntu:~/Documents/Tools/PyRep$ ll

I Can NOT find the file "_sim_cffi". 
Can anyone help me out with this? 
Thank you so much!!!

ffletcherr, 2021-05-17:
https://github.com/stepjam/PyRep#troubleshooting
If you are getting this error, then please check that you are not running the interpreter from the project root. If you are, then your Python interpreter will try to import those files rather the installed files.

deepdad, 2021-07-17:
Aside from what StepJam says, you can also `pip install -U cffi` (I don't think the version matters for cffi). It may be necessary to run `apt-get install libffi7 libffi-dev`. When installing PyRep it will create (compile) an interface with CoppeliaSim using ffi (Foreign Function Interface library runtime). The file name will be platform dependent but called _sim_cffi in the error message.

closed 2021-09-22 by stale[bot]

Wang-feihong, 2023-10-03:
> 除了StepJam所说的之外，你也可以`pip install -U cffi`（我认为版本对cffi来说并不重要）。可能需要运行`apt-get install libffi7 libffi-dev`。安装 PyRep 时，它将使用 ffi（外部函数接口库运行时）创建（编译）与 CoppeliaSim 的接口。文件名取决于平台，但在错误消息中称为 _sim_cffi。

> 除了StepJam所说的之外，你也可以`pip install -U cffi`（我认为版本对cffi来说并不重要）。可能需要运行`apt-get install libffi7 libffi-dev`。安装PyRep时，将使用ffi（外部函数接口库运行时）与CoppeliaSim创建（编译）的接口。文件名依赖于平台，但在错误消息中称为_sim_cffi。

安装上述命令后，也不能对应到这个_sim_cffi模块，请问如何修改？

55Dupup, 2024-12-19:
Hi, I also encountered this problem, how did you solve it? Looking forward to your generous reply!